prusa3d/Prusa-Firmware-Buddy #4050: [BUG] Prusa XL crashs/reboots when playing sound from file

foorschtbar, 2024-06-22:
### Printer model

XL 5T

### Firmware version

6.0.1

### Printing from...

Octoprint

### Describe the bug

When "printing" the attached gcode (list of M300 commands), my XL crashes and reboots to a "blue screen".

I tried the fix in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1075#issuecomment-1955316303 

[image]

### Files

[scooter.zip](https://github.com/user-attachments/files/15936725/scooter.zip)

Crash dump can provided if needed.

danopernis, 2024-06-26:
Thanks for the report @foorschtbar 

No need to provide crash dump, it is 100% reproducible. I have created internal ticket for investigation. Seems like MK4 and MINI are not affected (although it doesn't correctly apply tone durations, because the fix was not picked for the patch versions)

danopernis, 2024-06-27:
@foorschtbar The crash is caused by dividing by zero. Fix is scheduled for next patch release.

Note that even after the fix, your Scooter tune (which is really nice by the way :slightly_smiling_face:) won't play correctly. We would like to eventually fix that as well, but the fix is more involved, so we are not pushing it for the patch release.

To add more context, this boils down to the same underlying issue as https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3875 XL uses software PWM at 1kHz for managing its beeper, which is simply too slow to handle some frequencies and also dependent on actual CPU utilization.

danopernis, 2024-07-19:
@foorschtbar New firmware [v6.0.4](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.4) was just released which should fix the crash. Note that sound frequencies will unfortunately still be wrong.

closed 2024-08-08 by danopernis