prusa3d/Prusa-Firmware-Buddy #4051: [FEATURE REQUEST] Purge more menu option while purging

hackwin, 2024-06-22:
### Printer model

MK4

### Describe the feature

A purge happens to change filaments.  It takes one or more purges to load a new material/color.  The operator waits about 15 seconds per purge to select "Purge again" until they reach the desired extrusion.

The feature is to have the "Purge More" menu option be selectable while the printer is purging.  The purpose is to queue the purge requests and save on time waiting idle next to the printer.

### Expected functionality

A user is switching filaments from red to white.  It takes 6 purges and 1min 30 seconds until the extrusion is white, not pink.  The user selects "Purge More" several times while the first purge is occurring.  The purges requests are queued and run in series.  The user saves time from having to wait next to the printer for the color to be ready to print.

CZDanol, 2024-09-17:
Hello, we've assigned this request an internal ticket BFW-6068.

hackwin, 2024-09-19:
> Hello, we've assigned this request an internal ticket BFW-6068.

Maybe the 3d printer could purge until a stop button is pressed.  That way purging doesn't take several cycles of waiting.