prusa3d/Prusa-Firmware-Buddy #4052: [BUG] BSOD stubs.cpp:5 while receiving gcode from Prusa Connect

meermanr, 2024-06-23:
### Printer model

XL5T

### Firmware version

6.0.2+14873 [S][A]

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

Blue Screen of Death (BSoD) occurred shortly after sending a print job via PrusaSlicer-2.8.0-beta1 through Prusa Connect. I noted that the Prusa Connect tab in the slicer didn't seem to fully load (see screenshot) so I checked the XL's own screen and found it was still on the screen following "Change all filaments" I had done previously. So I used the swipe gesture for back and after the usual beep of acknowledgement no new menu screen was shown, instead there was a short delay and then this BSOD.

I suspect the bgcode was being streamed to the printer, or had recently finished being downloaded, when this happened.

BSOD text (using OCR, image also attached):

```
BSOD

stubs.cpp:5
aborted
task: displayTask
SP: 10007fd0 LR: 0802b15f PC: 0802b15e
CPUID: 0443c803 ICSR. ICSR_: 00020200 VTOR_: Fa050300
SCR _: 00000210 SHCSR: 00010000 DFSR_: e000ed34
AFSR: 00000030 ADR_ :00100030
bot: 100080bc top: 10006ab0 stack:
2000c7a8 00013391 00013391 40000c00 000c000e 00013391
10008000 00000003 00ffffff 00000000 00000030
00000000 00000000 00000000 08112234 0812Fddc
00000000 11000030 00000000 00000000 00000000
08112e34 0812fdcc 00000000 10000034 10000034 00000000
6.0.2+14873 [S] [A]
```

[image]

[image]

### Files

[PrusaXL5T_BSOD_stubs5.zip](https://github.com/user-attachments/files/15944584/PrusaXL5T_BSOD_stubs5.zip)

I will send the crash dump via email.

meermanr, 2024-06-23:
Crash dump sent to reports@prusa3d.com, and given ID: PR29049909378

danopernis, 2024-06-26:
@meermanr Thank you for the bug report and especially for the crash dump. I really appreciate including the report ID which helped speed up the process.

I acknowledge that the bug is present. It should be quite rare, but it is there nevertheless. For interested reader, it happens in [DialogHandler](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/v6.0.2/src/gui/dialogs/DialogHandler.cpp#L290) which didn't account for different [FSM generation number](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/v6.0.2/src/common/fsm_states.hpp#L20).

Fix is scheduled for the next patch release.

meermanr, 2024-06-26:
Hi Dano,

Thank you for acknowledging the crash, and the links to the code - it is
always interesting to read through the firmware. (Last time I was looking
to see if gcode are used under the hood by the filament load and unload
procedures, since I'd quite like to recompile the firmware and change the
filament temperatures, or add a new filament type - PHA).

Robert

danopernis, 2024-07-19:
@meermanr New firmware [v6.0.4](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.4) was just released which should fix this.

closed 2024-08-08 by danopernis