prusa3d/Prusa-Firmware-Buddy #4053: [ENHANCEMENT]  starting a print and showing options with the tool heads

Peewee0512, 2024-06-23:
### Printer model

XL 5 tool heads 

### Describe the enhancement

Ok so when you get ready to start a print and it show pla and on the first tool head and what not. I feel we should be able to pick a tool head that we want to print with. I don’t wanna unload tool head 3 filament to 1 just to use that filament

Tomcus, 2024-06-24:
Hi, I am not 100% what are you referring to, but it looks to me you are looking for tool mapping feature, that is already implemented: https://cdn.help.prusa3d.com/article/tools-mapping-xl_668277. Please ether verify that this is what you want or please be more specific about your feature.

Peewee0512, 2024-06-24:
Oh ok I’m sorry Sent from my iPhone

closed 2024-06-26 by danopernis