prusa3d/Prusa-Firmware-Buddy #4055: [ENHANCEMENT] Toggle STEALTH Mode in Prusa Connect

TimetravelerDD, 2024-06-24:
### Printer model

Prusa Connect

### Describe the enhancement

Description:
I propose adding an option in the Prusa Connect web interface to toggle STEALTH mode on or off. This mode, already implemented in the firmware, reduces noise during prints by adjusting travel speed and acceleration.

Details:

Feature: Checkbox or switch to enable/disable STEALTH mode.
Settings: Travel speed set to 80 mm/s and travel acceleration to 2500 mm/s² when STEALTH mode is on.
Benefit: Enhances user convenience for quiet nighttime printing without manual adjustments on the printer it self. 

[Link to original issue](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3804)

### Expected functionality

Since Printspeed etc. can already controlled in Prusa Connect it would be logical to have a stealthmode toggle there as well .

SusieKyuu, 2025-03-09:
I'd be interested to see this, too.

CZDanol, 2025-03-10:
Internal ticket: BFW-6889