prusa3d/Prusa-Firmware-Buddy #4058: Prusalink status isnt updating from previous state to "ready" in home assistant

Zero-Circle, 2024-06-25:
### Printer model

MK4

### Firmware version

6.0.2+14873

### Printing from...

Prusa Connect

### Describe the bug

Prusalink status isnt updating from previous state to "ready" in home assistant

### How to reproduce

Set the printer as ready.

### Expected behavior

When printers state changes to ready this should be visable in home assistant

Zero-Circle, 2024-06-25:
Please note that this issue persists across FW versions 6.0.1, 6.0.2 & 6.0.3

danopernis, 2024-06-25:
What is home assistant? Is this some third party application?

3d-gussner, 2024-06-25:
@danopernis Home assistant is great home automation platform. https://www.home-assistant.io/

3d-gussner, 2024-06-25:
@Zero-Circle Please check home assistant plug-in.
I assume that the Ready Not Ready are working correctly in PrusaConnect and PrusaLink but you only have issues with the home assistant plug-in.

Zero-Circle, 2024-06-25:
> @Zero-Circle Please check home assistant plug-in. I assume that the Ready Not Ready are working correctly in PrusaConnect and PrusaLink but you only have issues with the home assistant plug-in. https://www.home-assistant.io/integrations/prusalink/
> 
> In case the PrusaConnect and PrusaLink are working correctly you should close the issue here and open an issue at the home assistant plugin as this a third party product.
> 
> https://github.com/home-assistant/core/tree/dev/homeassistant/components/prusalink
> 
> https://github.com/home-assistant/core/issues?q=is%3Aissue+prusa

Is there a way for a noob to check the raw output of prusalink? SSH or such?

3d-gussner, 2024-06-26:
@Zero-Circle This is what I have found quickly as I am not that familiar with home assistant yet. https://github.com/home-assistant/core/issues/107242
Hope that helps. Please address your issues in their github repository and close it here.

Skaronator, 2024-06-27:
@Zero-Circle would be better to link the issues the other way around.

@danopernis the issue is that the printer doesn't report the Ready status. It doesn't exist as you can see here: https://github.com/prusa3d/Prusa-Link-Web/blob/ae51b7222c5865f6c796095f8251be05c0fce1f2/spec/openapi.yaml#L1269

I made an issue last year (#3640) requesting this info as well as allow setting the printer READY / NOT_READY.

danopernis, 2024-06-27:
@Skaronator oh good, thank you for the info. I will try to start the company wheels turning, now that I know what is actually being requested :slightly_smiling_face:

closed 2024-06-27 by danopernis

RacoJo, 2025-01-06:
Have you found a way to set the printer ready via HomeAssistant ?
I wanted a easier solution than playing with selenium via a python script