prusa3d/Prusa-Firmware-Buddy #4059: [BUG] MK3.5 6.0.3 Firmware - Print not centered on bed.

artistassembly, 2024-06-27:
### Printer model

MK3.5

### Firmware version

6.0.3

### Upgrades and modifications

Silicone Mod on Bed

### Printing from...

USB Drive

### Describe the bug

A print which is centered on the bed in Prusa Slicer 2.7.4 is offset to the right & not centered on the print sheet.  This can cause prints to fail as the printer attempts to extrude off the right side of the print sheet without any warning from the slicer.  The print also appears to be shifted to the front of the bed a similar amount, but there is enough clearance front to back to not cause a print failure.

I suspect this misalignment is also present in the application of the bed leveling mesh causing the compensation for the distance from the print sheet to the nozzle to be shifted as well. 

The print sheet is aligned accurately on the bed with the rear of the sheet against the 2 posts on the bed and the left & right sides of the sheet even with the sides of the bed.  This distance of the white "print area lines on this sheet are exactly 250mm x 210mm.

This issue was not present on the MK3S+ before the MK3.5 upgrade.  I'm seeing other users report this issue with their MK3.5 printers on Reddit as well as in this bug report https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3851.  

[MK3.5 Bed Offset Test.zip](https://github.com/user-attachments/files/16006538/MK3.5.Bed.Offset.Test.zip)

Overview 
[image]

Right Side
[image]
[image]

Left Side
[image]
[image]

Front
[image]
[image]

Rear
[image]

### How to reproduce

I have attached a zip file with a test stl which measures 249mm x 209mm as well as the 3mf & .bgcode files.  Printing this model with the stock Prusa Slicer 2.7.4 MK3.5 Printer settings should produce the same result.

### Expected behavior

I would expect the print to be centered on the print sheet with greater accuracy.

BlueFyre, 2024-06-27:
Thanks for filing this. Would like to see this acknowledged/addressed as well

MarkusReichart, 2024-06-27:
I also have this issue. While upgrading to MK3.5 I changed from nylon lock to the new spacers for the heat bed. 
Whether the old y-carriage in combination with new spacers cause this shift, or the new homing sequence set a different 0,0 position I don`t know. In #3851 they just limit the right boundaray -5mm for PINDA. Maybe it would be better to shift the whole origin more to the left.
Meantime I define offsets in PS:

[image]

nickduncan7, 2024-06-27:
I understand why Prusa says it should not be perfectly centered in the prior issue, and frankly that is not my expectation. However, if it was closer to center on the build plate, I believe the PINDA probing off the edge issue would have likely never been a problem.

Prusa-Support, 2024-08-08:
Thanks for reporting. The shifted MBL probing grid and the shifted print area are not necessarily related so the issues should ideally be listed separately.

This should not prevent users from printing large prints and the little tweak via Prusa Slicer printer profile suggested above is surely a valid option for the time being (optionally). However, we would appreciate seeing your measured offset so that we may eventually make more considerations on this supported by numbers - the more the tested printer, the better of course. 🙂 

Michele Moramarco
Prusa Research

nickduncan7, 2024-08-08:
> Thanks for reporting. The shifted MBL probing grid and the shifted print area are not necessarily related so the issues should ideally be listed separately.

The first sentence of the report in BFW-5606 is “The print area on the MK3.5 is not centered in the indicated print area on Prusa steel sheets.”

The first sentence of this report is “A print which is centered on the bed in Prusa Slicer 2.7.4 is offset to the right & not centered on the print sheet.”

I understand these are tracked as different causes internally, but from an external perspective the same issue has been reported twice and is ultimately still not resolved. The PINDA probing issue was mentioned solely as a side effect from the prior problem.

timur-tabi, 2024-08-12:
In the Bed Shape settings, what number should I put for the X Origin?

timur-tabi, 2024-11-04:
It's not stale, Prusa developers are just not interested in fixing their bugs. 

This didn't happen on the MK3.

timur-tabi, 2024-11-23:
@Prusa-Support  Any change of this being fixed any time soon?  I cannot print a 250mm-wide object because the extruder head keeps crashing into the right side.

Prusa-Support, 2024-12-07:
To my understanding, 250x210 mm prints were possible even if not centered.
I must be missing something but please note that an acceptable skew up to 0.25° may reduce the maximum axis length by up to 1mm or so.
https://help.prusa3d.com/article/xyz-calibration-details_2272
However, as of now, I can't recall or check if the exact same XYZ Calibration typical of MK3 is ported to MK3.5, including skew measurement.

I will report and review with the developers.
Nonetheless, more measurements and tests across more MK3.5 units may help.

Michele Moramarco
Prusa Research

ttabi, 2024-12-07:
Fair enough.  My only suggestion is that it should be possible to specify X/Y offsets in the Bed Shape setting and still see the Prusa bed image in the Plater panel.  As soon as I specify a non-zero offset, that image is replaced with a generic rectangle.

closed 2025-02-13 by github-actions[bot]