prusa3d/Prusa-Firmware-Buddy #4060: XL on 6.0.3 layer shifts

gbzh01, 2024-06-27:
### Printer model

XL 5 tool

### Firmware version

6.0.3

### Printing from...

Prusa Connect

### Describe the bug

Two printers, two different files, both ended up with layer shifts

danopernis, 2024-06-27:
@gbzh01 while I appreciate that you created an issue here, I can't fix the bug without having a clear reproducer. The info you provided is just not sufficient for any meaningful work to be done by firmware developers.

Please, talk to the support at https://help.prusa3d.com/article/customer-support_2287 and try to rule out hardware problems first.

closed 2024-06-27 by danopernis

gbzh01, 2024-06-27:
Hi

I am sorry to be so sloppy here but there is nothing I can do further. I upgraded yesterday to 6.0.3.

Result
2 XL machines
2 different files
layer shifts with each machine. I don’t see a reproducer. I just see layer shifts.

Never mind, I just downgraded back to 6.0.2 and keep on printing. Have a great day

Best regards,

Gregor Buehrer							
Director
___________________________

￼

		
Sputnique Creative GmbH
Industriestrasse 44
8152 Glattbrugg 
Switzerland

+41 76 586 04 83
www.SPTNQ.com