prusa3d/Prusa-Firmware-Buddy #4062: [BUG]  Unable to load Filament for coldpull with activated MMU

schorsch3000, 2024-06-29:
### Printer model

MK4

### Firmware version

6.0.3+14902

### Upgrades and modifications

MMU3

### Printing from...

not relevant

### Describe the bug

With no Filament loaded select
-> Control
-> cold pull
when asked to select what to to about filament select
-> load

the "Waiting for temperature" screen will just not end, the nozzle temperature stays at 215/215 for minutes. 

### How to reproduce

With no Filament loaded select
-> Control
-> cold pull
when asked to select what to to about filament select
-> load

### Expected behavior

after the temperature is reached i expect the process to go on

mody, 2024-07-03:
Thank you for the report. We've created an internal ticket with id BFW-5756.

aknezevi-work, 2024-07-19:
The rest of the cold pull menu works fine. The unload and the continue, if you successfully load the filament manually.

danopernis, 2024-08-26:
This should be fixed in v6.1.2 can you re-test please?

schorsch3000, 2024-08-26:
Can confirm, everything is fine now :-)