prusa3d/Prusa-Firmware-Buddy #4063: [BUG] Prusa Connect doesn't detect MMU3 after printer (re)boot

firetech, 2024-06-30:
### Printer model

MK4, with MMU3

### Firmware version

6.0.3 (also seen in at least 6.0.1 and 6.0.2)

### Upgrades and modifications

MMU3

### Printing from...

Prusa Connect

### Describe the bug

Every time I cold boot or reboot my MK4, Prusa Connect shows the printer as a stock MK4, without MMU3, even though the MMU3 is working perfectly fine. If I manually trigger a reconnect to Prusa Connect by either toggling the default network interface back and forth or toggling the Enable setting for Prusa Connect back and forth, Prusa Connect shows the printer as a MK4 with MMU3 after just a few seconds (toggling the MMU Unit off and back on has no effect). If nothing is done, Prusa Connect will continue showing the printer as a non-MMU3 MK4 through at least several prints.

It seems to me that the initial handshake with Prusa Connect is made too early, before the MMU3 has finished booting, so the MMU3 data isn't sent. Then, after the MMU3 has booted, the relevant data is never updated, unless the printer somehow loses connection to Prusa Connect.

This is especially annoying when using the new Prusa Connect integration in Prusa Slicer 2.8.0, as the printer isn't considered compatible with MK4ISMMU3 G-codes ("Printer model does not match") until Prusa Connect has detected the MMU3.

### How to reproduce

1. (Re)boot the printer
2. Wait for Prusa Connect to detect the printer as idle
3. Notice that Prusa Connect isn't showing the MMU3 (also not the Slots item in the status bar)

Workaround:

4. Go to Settings (or Tune, if already printing) > Network > Default and press the knob twice (to toggle back and forth between Eth and Wi-Fi).
5. Wait for Prusa Connect to show the MMU3.

### Expected behavior

Prusa Connect should notice the printer has an MMU3 connected after printer (re)boot without manual intervention.

cozmo14047, 2024-07-12:
And if the printer is off prusa connect doesnt show mmu so when you attempt to upload to prusa connect when the printer is off it says incompatable model

Krosta8, 2024-07-15:
I have the same issue as described. Someone on Discord told me to try reset on the printer, but that did not help at all. Which make sense regarding what firetech wrote. If I reset the printer, MMU3 is fully booted too late for Prusa Connect to recognized connected MMU3. I can not use new feature of PrusaSlicer 2.8.0 as well, because of this :-(

atkinsonaaron, 2024-07-20:
Same issue - the MK4 printer doesn't recognize the MMU3 is installed.
This occurred with Firmware 6.0.4.  The MMU3 stopped working at firmware 6.0.3

cozmo14047, 2024-07-21:
I belive the issue is due to being able to turn the mmu off, please do fix this.

danopernis, 2024-07-23:
Thanks for reporting. We have a fix for that scheduled for next release. The issue is caused by race condition when printer connects to Prusa Connect before MMU boots.

danopernis, 2024-08-26:
This should be fixed in v6.1.2 via https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/fc1b163a0e6e32ee0b9f4a0c29df6bb6b885c31b can you re-test please?

Krosta8, 2024-08-26:
Yesterday I did a print and it was working correctly. Just in Prusa App I see it like without MMU (not sure if the app can show with/without MMU variant)

cozmo14047, 2024-08-26:
Fixed for me but like Krosta8 said in the app the image is just a standered mk3.5 but on the web it shows correctly

closed 2024-11-05 by github-actions[bot]

RobiFish, 2025-04-08:
I got the same problem, but with my MK3S+ (2 of them). I Can´t make them work as MK3S+MMU3. 
When I shut on the printer, Prusa Connect says: Yes, MMU3, all fine, but printer is busy. After reset he's not busy anymore, but lost his MMU3 in Connect. I use 3.14.1 E3DREVO FW on printers and 3.0.3 on MMU.
Connection between printer and Prusa Connect is via Pi Zero GPIO. 

Any ideas? Thank you very much!

MxClaudio, 2025-04-22:
I have the same problem, I think that there is no longer any desire to fix the problems that have been created.

CZDanol, 2025-04-22:
You are writing comments to an issue that has been reported fixed by the users and closed. If you are experiencing the issue again, please create a new ticket.

MxClaudio, 2025-04-22:
all the latest firmware, 3.14.1 for mk3s+ and 3.0.3+896 per mmu3. And the problem has not been fixed at all in the official firmware. The printer, when turned on, never gets to the ready state. In addition, Prusa Connect does not recognize the printer as a mmu3

firetech, 2025-04-22:
> And the problem has not been fixed at all in the official firmware.

Yes, it has. You're barking up the wrong tree.

This is the repository for the MK4(S)/Core One/Mini/XL firmware, which is entirely separate from the firmware in MK3S+ and older i3 models. Since you have an issue with MK3S+, you should report your issue here instead: https://github.com/prusa3d/Prusa-Firmware. The problem you're experiencing on MK3S+ can therefore not be _the same_ issue issue as this (on MK4 there was never any issue with the printer being marked Busy), but it is indeed _similar_.

3d-gussner, 2025-08-11:
@RobiFish Please contact PrusaSupport if you still have the issue. It is a know issue with PrusaLink for RPi

pontab, 2026-04-25:
[image]

[image]

I cant find the MMU in prusa connect, havent used it for a while but I think it was working a year ago. Now I have updated the firmware on both the MK4 and the MMU3, but prusa connect still dont show it.
The workaround suggested here does not fix it for me :/