prusa3d/Prusa-Firmware-Buddy #4064: Stop the blinking, please....

BrienKing, 2024-07-01:
### Printer model

XL Multi Tool Head

### Describe the enhancement

Right now the LED's on the extruders just sit there and blink for no reason.   Can we have those OFF if they are not being used.  If they are used they should be a SOLID color.   Green for ready, White for In use.  Red for an Error,  off if not being used.

### Expected functionality

"We've all got our switches, lights, and knobs to deal with, Striker. I mean, down here there are literally hundreds and thousands of blinking, beeping, and flashing lights, blinking and beeping and flashing - they're *flashing* and they're *beeping*. I can't stand it anymore!"  -- Commander Buck Murdock

tg73, 2024-07-02:
My guess is that the blinking gives confidence that the MCU's main loop is ticking and has not crashed. I'm not bothered by the flashing, but I sympathise with your preference.

BrienKing, 2024-07-02:
That is possible, however, they could move that to a diagnostics menu.

KlausKraemer, 2024-08-04:
I am quite disappointed about the toolheads' LEDs. They don't give any useful information.

There should be information about statuses:
- heating
- cooling
- standby
- printing (now blinking white, which is ok for me)
- cold (to safely handle manual work on the tool)

Currently the LEDs seem quite an overkill for the little information they provide.

legomind, 2024-09-09:
Would love the leds on the toolheads to be more informative, as well as the option to turn them of entirely, as well as only lighting the toolheads that are being used in the current print

legomind, 2025-11-26:
Am I the only one who has all white LEDs on my xl tool heads, all the time?

None of my tool heads have EVER been colored.

KlausKraemer, 2026-09-28:
No! The problem is still there, but nobody bothers to resolve it. Even, when safety could be improved by a reasonable use of the LEDs.

legomind, 2026-09-28:
keep open