prusa3d/Prusa-Firmware-Buddy #4068: Prusa XL printer identifies itself as MK4 over WIFI[BUG]

langph, 2024-07-02:
### Printer model

XL multi 5 heads

### Firmware version

6.0.3

### Upgrades and modifications

None

### Printing from...

Prusa Link

### Describe the bug

The WIFI name of the XL shows as MK4 in my client list. LOL. I cannot change the name in the printer.

### How to reproduce

Activate WIFI, add it to a network, look at all WIFI clients in the router..

### Expected behavior

The WIFI client name is "Prusa XL", or "The Mighty One" or "The 5 Headed Prusa".

langph, 2024-07-03:
[image]

bryn51, 2024-07-08:
Adding the ability for the Network Identified name of the Printer device would be a great enhancement for all the Prusa Printers.  It could be implemented by adding a Printer Name to the settings.ini file, alongside the wifi credentials.

murk-sy, 2024-07-15:
This is a Unifi issue, not related to the firmware. Unifi identifies device manufacturer (and type) by its MAC address.

It seems someone submitted just about every MAC address prusa used to be a MK4 (which it's not).

Click on the icon and then click "No icon available".

Before
[image]

After
[image]

closed 2024-09-24 by github-actions[bot]