prusa3d/Prusa-Firmware-Buddy #4069: [FEATURE REQUEST] The ability to ready from print summary page

andgju, 2024-07-02:
### Printer model

MK4, MK3.9, MK3.5, MINI & XL

### Describe the feature

The ability to set the printer status to ready from the post-print screen. Instead of going into [Control>Set Ready]

Like in the photo below, where I've replaced the greyed out "tune" button with a ready button. 
[image]
compared to the original:
[image]

CZDanol, 2024-07-04:
Internal ticket: BFW-4938

bryn51, 2024-07-08:
I do not agree that implementing this request would be a good use of scarce resources.
I vote against this enhancement.,

andgju, 2024-07-08:
It would save a lot of clicks when using the queue function in Prusa Connect

CZDanol, 2024-07-09:
Some sort of solution like this is in our plans :) Can't promise you when though

bryn51, 2024-07-09:
> Some sort of solution like this is in our plans :) Can't promise you when though

The process used by prusa in deciding what work to do on software development is totally opaque. I wonder who they ask if a particular new feature is welcome of unwelcome, indeed what are the priorities and development roadmap. Certainly I've not seen anything like a survey put out to customers.
Whats strange is that in recent times we see new features released before they are ready for prime time: they come out half baked and not fully functional, and with almost zero documentation on usage. Release notes incomplete, user guide not updated etc.
Is this a product of the fact this is not a paid product and in the public domain ?
Whatever the case I observe the folks coming on here are arrogant in the extreme and appear to be faceless and  unaccountable.

dillenburg, 2024-07-10:
I would love this soloution, it would save me a lot of time

BugsBunny1403, 2024-08-08:
Nice idea

Prusa-Support, 2024-10-11:
Thanks for the suggestion.
The "set ready" option is available right in the "print finish" screen from now with [FW 6.2.0-alpha2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-alpha2).

Michele Moramarco
Prusa Research

closed 2024-10-11 by Prusa-Support