prusa3d/Prusa-Firmware-Buddy #4070: [ENHANCEMENT] Add option to disable serial printing screen

PinkPandaKatie, 2024-07-03:
### Printer model

All models

### Describe the enhancement

I use Octoprint solely as a means of external control. I have a USB number pad next to my printer which I have connected to a laptop, which runs a program that sends GCode commands via Octoprint when keys are pressed. This allows me to quickly set temperatures, move the head around, control the fans, etc.

The annoying thing is, though, any codes that move the head cause it to enter "serial printing" mode, blocking access to the UI. Worse, there have been a few times where I unloaded the filament but didn't pull it out of the sensor right away, then I moved the head up with the numpad, and pulling the filament out triggered a filament runout sequence that I could not cancel without resetting the printer.

### Expected functionality

Two possible solutions:

* Add a toggle setting in "User Interface" to enable/disable the serial printing screen. SerialPrinting::serial_command_hook can check this setting and do nothing if it is disabled.
* Add a GCode toggle that Octoprint could send which would indicate that following commands are not part of a print, but are just control moves. This would be more complex and require allocating and documenting the GCode as well as an enhancement to Octoprint.

I have implemented the first option myself and can submit a PR for it.

PinkPandaKatie, 2024-07-07:
PR is here: https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/4071

I expected it to auto-link back but it didn't

danopernis, 2024-08-15:
Merged. This is not a bugfix, so I expect this to be released in the next non-patch version. Keeping the issue open to track progress.

closed 2025-12-09 by github-actions[bot]