prusa3d/Prusa-Firmware-Buddy #4073: [BUG] Dead-end while "First Layer Calibration" on a new sheet

MarkusReichart, 2024-07-03:
### Printer model

MK3.5 MMU3

### Firmware version

6.0.3

### Upgrades and modifications

MMU3

### Printing from...

USB

### Describe the bug

"First Layer Calibration" for a new sheet ends in a dead-end.

### How to reproduce

MMU3 active. No filament loaded. Navigate to Hardware/Settings/Steel Sheets/ and choose a "configured" such as "Custom4".
Select "First Layer Calibration" and choose **"LOAD"**. Select PETG and wait until heat up is finished. Progress stop at 16%. Dead-end. 

### Expected behavior

No dead-end for First Layer Calibration choosing LOAD. If I choose "UNLAOD" a not inserted filament the calibration process starts after the "unload of nothing"

### Files

[image]

MarkusReichart, 2024-07-03:
Above "in how to reproduce" a important word is missing: choose a "**"non** configured" (sheet). 
Become aware of my typo I navigated to "Filament" and UNLOAD the filament. Choose First Layer Calibration on a "configured sheet" and now get a dead-end at 0% Parking. "Stop"button is not responsive.
[image]

antimix, 2024-07-06:
PRINTER:  MK3.5 REVO 40W MMU3 FW 6.0.2
It happens normally also with configured sheets.

1. Ensure there is no filament in MMU3 or Nozzle
2. Select First Layer Calibration
3. Select **LOAD**
4. Select PLA and wait for the temperature reached.
5. You will notice that the while the Nozzle temp is set, the BED temperature is not set!
6. The printer waits forever.

[image]

7. The bed will stay at environment temperature
8. No life from the MMU3
9. Moving the knob does nothing
10. Only the RESET button can unblock the situation.

danopernis, 2024-07-08:
Thanks for reporting @antimix @MarkusReichart 

I have created internal ticket for investigation.

danopernis, 2024-07-16:
Fix is scheduled for the next patch version. Calibration was stuck because the MMU slot was unspecified when the filament was unloaded before starting calibration.

danopernis, 2024-07-19:
@MarkusReichart @antimix New firmware [v6.0.4](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.4) was just released which should fix this.

MarkusReichart, 2024-07-20:
Does work as expected now. Thanks.