prusa3d/Prusa-Firmware-Buddy #4076: [BUG] MK3.5 heatbed selftest error

rafalr100, 2024-07-04:
### Printer model

MK3.5

### Firmware version

6.0.3

### Upgrades and modifications

None

### Printing from...

USB drive

### Describe the bug

After the upgrade from MK3s+ to MK3.5 I was running a selftest. During a selftest all test are passed except the heatbed which is failing. I was discussing this on a chat with many Prusa technical guys and checked all the possible components but all the results were correct (voltage, resistance etc.). I even replaced a thermistor but the error still exists. It also exists in older firmware version, even below 6.0.

One of the Prusa tech guy mentioned that this may be a firmware issue so I'm raising this for your feedback. 

Right now every time I start a printer and before each print I'm getting an error that a testing and calibration has not been done (only heatbed test is failing). I can start and successfuly complete a print, this error is not stopping me from printing. Heating bed is functioning correctly and heating up correctly.

During a self test it heats up to around 70 degrees, then a test time comes to end and is showing a red X.

rafalr100, 2024-07-04:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Prusa-Support, 2024-07-13:
Hello. Not sure what may have gone wrong in the troubleshooting session with Tech Support but something is clearly wrong on the hardware side.
Feel free to write to my attention - my name is in the signature (info@prusa3d.com) - so I can review, correct and report any negligence on our side if that is the case.

The heatbed test would hardly go beyond ∼60°C during the test, and for a large range of conditions, **unless the spring sheet is missing**. The test would be already doomed to fail if the heatbed reaches ∼65°C before the progress bar fills up.

So **make sure to put the spring sheet on the heatbed** and, if the test still fails, we may be looking at very specific environmental conditions or something wrong with the printer.

Your heatbed reached 70°C so the firmware is doing it right and this issue could be closed as non-firmware related.

Michele Moramarco
Prusa Research

geekcouplettv, 2024-07-17:
I have the same issue but with MK4 on firmware 6.0.1 and 6.0.3.
I have two printers.

I wanted to check on one printer heatbed self test and hotend test because My old termistor on hotend end it's life.
And after installing it i went to calibration and check if it okay.

First on that printer where i changed the themistor i had fail because of hotbead was only going to 70 degress and then fails.

And I tested on my 2nd MK4 Printer witch is on 6.0.3 firmware. 

The 2nd printer some time ago went all self test without problem but now it is also saying error in heatbed selftest. It dosent go beyond 70 degress and it is showing fail.

So for me there is a problem with firmware not hardware. Printers are printing without problem PLA/PETG/ASA/PC

But cannot pass the selftest

Kikkawaryu, 2024-07-17:
I had similar issue when upgrading to MK3.5

I took the Kapton tape off, ensured the bed thermistor was in the dead centre of the circle and orientated correctly it should be flat side down with the wires facing towards, put both sets of tape back down pressing all around the thermistor to create a good seal 

It now self tests with no issues.

closed 2024-09-24 by github-actions[bot]