prusa3d/Prusa-Firmware-Buddy #4077: [BUG] [BSOD] stubs.cpp: 5 MK4 crashed when I powered off the OctoPi

AxisRay, 2024-07-04:
### Printer model

MK4

### Firmware version

6.0.3

### Upgrades and modifications

A DC-DC module is connected to the 24V power adapter of the MK4, supplying power to the Raspberry Pi

### Printing from...

USB Drive (not from octoprint)

### Describe the bug

[image]

### How to reproduce

Connect the octopi to the MK4, then cut off the power of the octopi.
(I am not sure if it can be reproduced.)

### Expected behavior

Nothing happen

### Files

The crashdump has been sent to reports@prusa3d.com.

AxisRay, 2024-07-04:
PR29192035778

danopernis, 2024-07-08:
Hello @AxisRay thanks for the report. Upon examining the provided crash dump, this turned out to be the exact same issue as https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4052 Fix is already scheduled for next patch release. FWIW, this has nothing to do with powering off the octopi.

AxisRay, 2024-07-08:
@danopernis Thank you for the update and appreciate your prompt response.

danopernis, 2024-07-19:
@AxisRay New firmware [v6.0.4](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.4) was just released which should fix this.

closed 2024-08-08 by danopernis