prusa3d/Prusa-Firmware-Buddy #4078: [FEATURE REQUEST] Include all temperature sensors of the MK4 in PrusaLink telemetry data

Zappes, 2024-07-05:
### Printer model

MK4

### Describe the feature

As the owner of a Prusa Enclosure
I want to monitor the xBuddy board temperature
So that I can take measures when the temperature gets too high

Background

When printing high temperature filaments like PC Blend or Nylon in the enclosure, the temperature of the mainboard can get into ranges where proper functioning isn't guranateed anymore. I would like to be able to react to such a situation, e.g. by activating a cooling solution.

My ultimate goal is to control the behavior of my system using Home Assistant. The data source for that system is the [PrusaLInk Integration](https://www.home-assistant.io/integrations/prusalink) which, in turn, is based on the telemetry data from PrusaLink. For that reason, the first step towards my goal would be the inclusion of the board temperature in the telemetry data.

### Expected functionality

The expectation is that the telemetry data provided by PrusaLInk is extended so that the values of all sensors in the system are reported. At the very least, the MCU temperature should be there, but every bit of additional information would be valuable.

cozmo14047, 2024-07-10:
I second this.

thijstriemstra, 2024-07-23:
> When printing high temperature filaments like PC Blend or Nylon in the enclosure, the temperature of the mainboard can get into ranges where proper functioning isn't guranateed anymore. I would like to be able to react to such a situation, e.g. by activating a cooling solution.

Is this documented somewhere in official prusa documentation? And what are the symptoms? What temperature btw?