prusa3d/Prusa-Firmware-Buddy #4079: [ENHANCEMENT] MMU filament change not reported

deimosfr, 2024-07-05:
### Printer model

MK4 with MMU3

### Describe the enhancement

Hi,

When using MMU and filament requires to be replaced, the status is not correctly reported. Here is the output I have with `curl -H "X-Api-Key: xxx" http://prusa/api/v1/status | jq` command:
``` json
{
  "job": {
    "id": 418,
    "progress": 78.00,
    "time_remaining": 840,
    "time_printing": 3390
  },
  "storage": {
    "path": "/usb/",
    "name": "usb",
    "read_only": false
  },
  "printer": {
    "state": "PRINTING",
    "temp_bed": 60.0,
    "target_bed": 60.0,
    "temp_nozzle": 220.0,
    "target_nozzle": 220.0,
    "axis_z": 30.4,
    "flow": 100,
    "speed": 100,
    "fan_hotend": 8309,
    "fan_print": 6238
  }
}
```
The reported status should be "Attention", like without MMU.

Thanks

PinkPandaKatie, 2024-07-06:
I did a bit of digging, and it looks like this case is supposed to be handled, in `src/state/printer_state.cpp`:

[image]

`load_unload_attention_while_printing` looks for `MMU_ERRWaitingForUser`, which the `FILAMENT_EJECTED` error should set:

[image]

I` can't really debug it at the moment but this just happened to me so I can confirm it.

zeroflow, 2024-08-15:
I also ran into this bug / missing element yesterday.
Any MMU attention, e.g. not able to feed to Extruder, Filament Runout etc. should lead to a status change.

In my case, the filament ran out - which the printer detected and remedied - but the status didn't change and the printer stayed at "Printing".
This would be really convenient to have, e.g. to alarm you via HomeAssistant.

closed 2024-10-24 by github-actions[bot]