prusa3d/Prusa-Firmware-Buddy #4080: [BUG]  QUICK KNOB ROTATION CAUSES CRASH & REBOOT DURING FIRST LAYER CALIBRATION

antimix, 2024-07-06:
### Printer model

MK3.5

### Firmware version

6.0.2

### Upgrades and modifications

REVO 40W NOZZLE

### Printing from...

LCD local hardware function

### Describe the bug

During the first Layer Calibration, if the knob is not rotated very slowly the printer reboots.
I was forced to do the process at least 7 times, because always in the middle of the calibration it rebooted since I was not rotating slowly enough the knob.

### How to reproduce

1. Start the First Layer Calibration process
2. When it runs, rotating the knob it increases or decreases the distance from the nozzle to the bed (e.g. 0.000,-0.003,-0.004...)
3. Now you realize that you need to move at least to -1.780 and need to lots and lots of rotations.
4. Rotate quickly enough (or very quickly) the knob to be able to arrive to -1.780 before it finish the process.
5. As you increase the knob rotation speed immediately the process freeze, and the printer crashes and reboot.

### Expected behavior

Whatever is the knob rotation speed, the printer should not reboot, and it should have the same behaviour it has during the Axes move, where you can move very quickly the knob to move long axes distances.

danopernis, 2024-07-08:
Hello @antimix do you have the crash dump? Can you send it to reports@prusa3d.com and provide a reference number here?

antimix, 2024-07-08:
@danopernis No, I don't have. The printer does not have a crash screen, but it just reboots.
You may recreate your issue it on any MK3.5 6.0.2 just rotating quickly the knob during first layer calibration 😅
In the meantime I have upgraded to 6.0.3 and I will check if the issue is still present.

danopernis, 2024-07-08:
Well, I tried that and I am unable to reproduce it. But I have noticed that during vigorous knob-turning, entire xLCD panel is moving a lot. Might be just a hardware/electronic problem, when you are accidentally shorting the reset signal. Can you try that in other screens, not just the first layer calibration?

antimix, 2024-07-08:
@danopernis  I will, especially now on the 6.0.3. 
I normally rotate the know like a crazy when I move the axes, especially Z, and the printer never crashed.
It is strange you can rotate very quickly without a reset. What happen is that there are no movement or flickering on the LCD during rotation until I reach a certain speed, and the the screen freeze for 4 seconds and the printer reboots.
May be I can create a nice video 😁

ClaGre, 2024-07-14:
@antimix try to flash the firmware again, I'ved the same issue with 6.0.2 then with latest upgrade to 6.0.3 was fixed.
Moreover I've changed the speed with I rotate the knob without high speed rotations, because I've think that the fast values change maybe can confuse or saturate some buffer allocated for this operation 🤷🏻‍♂️