prusa3d/Prusa-Firmware-Buddy #4081: [ENHANCEMENT]  INCREASE THE FIRST LAYER CALIBRATION RANGO TO -3.000

antimix, 2024-07-06:
### Printer model

MK3.5 REVO

### Describe the enhancement

Currently the distance between the nozzle and the bed during the first layer calibration have a range of +2.000  <=< -2.000

### Expected functionality

With a E3D REVO nozzle , some time, the actual range may not be enough after installing a new nozzle.
It would be better to have the possibility to have +2.000 <=< -3.000

jlandegent, 2024-07-07:
i had the same issue. simple solution would be to place the pindaprobe further away from the bed. i just used a thicker tiewrap for calibration.

bryn51, 2024-07-08:
This looks like something that a user could handle themselves by adjusting the printer physical settings to cope with the fact they chose to deploy a different hotend than the standard.
I vote this enancement in the negative.

Levi--G, 2024-07-13:
This is definitely not just a custom hotend issue, i run into the same issue on a stock mini on the textured PEI sheet. I need -2.010 but it stops at -2. Its been this way since it came pre-assembled. So definitely not an end user issue!

danopernis, 2024-08-14:
Basically what @jlandegent said :+1: 

@Levi--G on MINI, you can follow [this guide](https://help.prusa3d.com/guide/height-adjustment-of-m-i-n-d-a-superpinda-sensor-mini-_154547) to adjust probe height.

@antimix I don't know how the REVO hotend looks like, but it seems to me that you can follow similar process on MK3.5

If you have any further questions, [feel free to contact our support](https://help.prusa3d.com/article/customer-support_2287). We are not planning to increase the live Z adjust range, 4 mm should already give you enough leeway. This is the same range as the 8-bit firmware for MK3 has.

closed 2024-08-14 by danopernis

Levi--G, 2024-08-14:
> Basically what @jlandegent said 👍
> 
> @Levi--G on MINI, you can follow [this guide](https://help.prusa3d.com/guide/height-adjustment-of-m-i-n-d-a-superpinda-sensor-mini-_154547) to adjust probe height.
> 
> @antimix I don't know how the REVO hotend looks like, but it seems to me that you can follow similar process on MK3.5
> 
> If you have any further questions, [feel free to contact our support](https://help.prusa3d.com/article/customer-support_2287). We are not planning to increase the live Z adjust range, 4 mm should already give you enough leeway. This is the same range as the 8-bit firmware for MK3 has.

And how am i supposed to calibrate the non textured sheet then? which is higher? Am i supposed to position the probe in the 0.5mm margin i will have? Really for something as simple as changing a single limit you are making this into a very big thing... As if anyone except those who need it will notice they can go up to 3mm now... I expected better from prusa...

antimix, 2024-08-14:
@danopernis from your comments on issue #4087  I understand that the first layer calibration code was written in a rather complex manner, and finding and changing the lower limit, may require too much efforts and can introduce bugs. So it is easier to physically raise the SUPERPINDA from the bed 1 mm more, and the calibration limit should be enough.
I have already used that trick on my MK3.5 and I was able to calibrate the first layer as I wanted.
For me this issue can be closed for now, since there are other more important issues open to work on....