prusa3d/Prusa-Firmware-Buddy #4082: [BUG] BSOD at print start with Apricorn encrypted usb drive

13jmb3, 2024-07-11:
### Printer model

XL Multi Tool

### Firmware version

6.0.3

### Upgrades and modifications

None

### Printing from...

USB Drive (Aegis Secure Key 3Z)

### Describe the bug

When starting a print using the [Aegis Secure Key](https://apricorn.com/aegis-secure-key-3z), the printer immediately crashes and shows the bootloader screen, followed by the BSOD Watchdog Reset screen.

Printing works from the supplied USB key, however I am required to use the Aegis Secure Key due to IT policy.

### How to reproduce

Start any print from gcode on the Aegis Secure Key

### Expected behavior

Print should continue after hitting "Print".

### Files

[image]

13jmb3, 2024-07-12:
> Hi @13jmb3 thanks for the report. To further investigate this bug, I need you to send a crash dump file to [reports@prusa3d.com](mailto:reports@prusa3d.com) referencing this issue. We do not have Aegis secure key in the office and I do not feel like purchasing one, so I really need that dump in order to help you with your issue.

Crash dump has been sent.

13jmb3, 2024-07-15:
I attempted the print with a separate-power USB cable and had the same crash. The USB remained unlocked the entire time with the new cable.

danopernis, 2024-07-16:
@13jmb3 we received your crash dump, but unfortunately I am not able to retrieve any meaningful info from it. Either the file is severely damaged, or it doesn't originate from XL 6.0.3. Now I don't think you've sent the wrong dump. This may indicate some massive memory corruption long before the printer actually crashed due to watchdog timeout.

What precisely do you mean by printing with separate-power USB cable and USB drive remaining unlocked? I find it quite hard to follow.

13jmb3, 2024-07-16:
@danopernis The USB drive requires a passcode to be entered on its keypad, after which it decrypts the data and is readable. If power to the drive is interrupted, the drive immediately encrypts itself again. The cable I tried is similar to [this](https://www.startech.com/en-ca/cables/usb2habmy3), where power to the drive is provided separately from the printer. This rules out momentary power-loss from causing the crash.

Should I send in additional crash dumps?

closed 2025-02-18 by 13jmb3

danopernis, 2025-02-18:
But how? 6.3.0 is released only for CORE One at the moment.

13jmb3, 2025-02-18:
> But how? 6.3.0 is released only for CORE One at the moment.

Should be 6.2.0, I read too many change notes before this