prusa3d/Prusa-Firmware-Buddy #4084: [BUG] "flow factor" in "tune" menu doesn't read correctly first printer start

Krulle86, 2024-07-12:
### Printer model

MK4

### Firmware version

6.0.3+14902

### Upgrades and modifications

None

### Printing from...

USB Cable/Octoprint

### Describe the bug

I noticed today that the "flow factor" is read in at the wrong time after starting the print.
Specifically: I switch on the printer, start printing, the printer has to heat up and I change the "flow factor" (100%) in the "tune" menu during the heat-up time, then this is not applied during printing and it prints with the default values. I have to confirm the value again during printing so that the flow factor is corrected. 
It seems to me that the flow factor is not applied during printing or is read at the wrong time. It is also only after the printer has been switched on directly.

### How to reproduce

Switch on the printer, start printing, the printer has to heat up and change the "flow factor" (100%) in the "tune" menu during the heat-up time. Then the new value is not applied during printing and it prints with the default values.
after that, you have to confirm the value again.

Krulle86, 2024-07-12:
Sorry, I think this ended up in the wrong repo. Can someone move it to the firmware repo? Thank you.

danopernis, 2024-07-12:
Does your G-code contain any M221 commands? Setting flow factor from tune menu behaves similarly to just doing M221 from G-code. There is only one flow factor variable in the code and it gets overwritten by last request to change it.

Krulle86, 2024-07-12:
> Does your G-code contain any M221 commands? Setting flow factor from tune menu behaves similarly to just doing M221 from G-code. There is only one flow factor variable in the code and it gets overwritten by last request to change it.

Thanks for your reply.
Yes, there is in front of the code a "M221 S100 ; set flow to 100%" (don't know where it comes from).
I interpreted the setting that the flow factor from the slicer is "corrected" or multiplied by this value. It's pretty pointless that way, especially if you use several different flow factors in the GCode (for whatever reasons).

danopernis, 2024-07-12:
M221 is set by the slicer, you can edit that in expert settings. But then you risk running new prints with flow factor of previous prints, which you may not want. While I agree this might be confusing, I don't consider this to be a bug, so I am closing this issue.

Maybe we could introduce two factors which would multiply, one controlled by the G-code and another controlled from the menu, but I can already foresee somebody wanting to setup the second factor from outside the menu (from Connect or Octoprint or anything else) and people having different expectations as to various corner cases.

closed 2024-07-12 by danopernis