prusa3d/Prusa-Firmware-Buddy #4086: Stuck Filament detection not working

bryn51, 2024-07-17:
### Printer model

Mk4

### Firmware version

6.0.3

### Upgrades and modifications

Prusa Original Enclosure

### Printing from...

Prusa Connect

### Describe the bug

During a print using Polymaker CoPA, a 0.25 mm nozzle, and the filament gets stuck in the hotend during the print.
I grant that a 0.4 mm nozzle would have been better, but high accuracy and resolution were called for on the model.
I come back after some time and find that the printer is moving in mid air, not extruding, and positioned well above the last layer printed.
The filament stuck in the nozzle condition was not detected, the print carried on, the printer unaware of this event, and the print was lost.

### How to reproduce

Execute a simple print using Polymaker CoPA on a 0.25 mm nozzle, or another filament that will not tolerate a nozzle too small.  It will jam sooner or later. Confirm if the stuck filament detection works as expected.

### Expected behavior

The stuck filament would be detected using the load cell sensor. If stuck filament occurs, the print should immediately stop, and:
- Maintain the bed temperature
- alert the user the fault has occured
- Offer the user the opportunity to clean the nozzle via "change filament" methodology
-  on completion, resume the print.

tlachmann, 2024-07-21:
Hello,

since a few days I,m struggeling with the same issue, in that case Esun PETG-HS Filament black.

Earlier (after first layer) or later (at approx 4mm height) the extruder stops moving the filament to the nozzle and no "filament stuck" notice is shown. (option is enabled)

After stopping the print, the filament will be always unloaded, so it does not jam in the extruder.

attached you can find pictures of the filament after unload. (It happens with less and more idler tension setting)
[image]
[image]
[image]
[image]

bryn51, 2024-07-21:
the good news is that Prusa Support have divulged that this is being worked on. But no indication when the work will be completed as yet. Encouraging nonetheless.

bryn51, 2024-07-21:
> Hello,
>
>
>
> since a few days I,m struggeling with the same issue, in that case Esun PETG-HS Filament black.
>
>
>
> Earlier (after first layer) or later (at approx 4mm height) the extruder stops moving the [...]

I'd suggest you have the two issues:
- why is the filament jamming ? possible your nozzle temp is on the low side (that has happened to me)
- the firmware issue reported here.

gwskala, 2024-07-23:
I'm not a programmer but thought I'd share my stuck filament error experience with my XL after installing the Prusa enclosure, upgrading some extruder parts and upgrading the XL firmware from V6.0.0 to V6.0.3.  Here's a link to my post (and others with this problem): https://forum.prusa3d.com/forum/postid/716679/

My issue was resolved by re-flashing back to V6.0.1.  I don't know what's changed but something is acting different in V6.0.3 (and some others suggest 6.0.2 may cause this problem)

cazdghost, 2024-08-08:
Encountering this with 6.0.4 FW for MK4 with MMU3. rollled back to 6.0.1 as suggested and currently testing on a 23hr print. It is at 60% now and still dont have nay Stuck filament error.

Rylan-Meilutis, 2024-08-11:
I had this issue and noticed that for me it only happened with certain brands of pla, my best guess is that the extruder area got hot enough that the pla softened and jammed, I ran the exact same gcode with a different pla and the print went without issue.

Nitrowing, 2024-08-29:
My hours old MK4S is doing this. All up to date firmware. Prusament black PETG. Sick of this already and going back to my faultless 6yo Mk2S - if this issue isn't sorted by the weekend the printer is getting boxed and returned.

makermelissa, 2024-09-23:
I just disabled the feature in my MK4S menu and now it's working perfect. Seems like it's too sensitive.

rob-miller, 2024-10-08:
Prusa XL, firmware 6.1.3, PETG, stuck filament detection enabled, enXlosure open with fan blowing into printer (warm environment).

I have repeated stuck filament events without being detected, resulting in 'air printing'.   I have the same damage to the filament from the extruder drive wheel as shown in the photos.

Regardless of the cause of the cause of the filament jam, the problem is that it is not detected - as described in **this** bug report.  **Please** open a different issue if you are getting spurious/inappropriate/excessive warnings about stuck filament issues.

cazdghost, 2024-10-09:
> Encountering this with 6.0.4 FW for MK4 with MMU3. rollled back to 6.0.1 as suggested and currently testing on a 23hr print. It is at 60% now and still dont have nay Stuck filament error.

Update on my issue with Stuck Filament.  when i downgraded to V6.0.1  it improves but later on went back and got much worse and have stuck filament on every print. I opened up the nextruder and found out a strip of filament(around1.5mm) stuck inside the planetary gear. which most likely causing the stuck filament error. After this havent encountered the error again.  Hope yours have the same cause and can fix it soon.

tlachmann, 2024-11-02:
If someone is interested how to reduce filament jam/stuck and the issues why it actually happens:

https://www.printables.com/model/1059347-mk4mk4s-with-mmu-main-plate-with-improovement-step

CZDanol, 2025-01-04:
Hey, the "stuck filament" feature is named a bit inappropriately - what it actually does is that it detects the E motor skipping. The filament can however get stuck in a way that does not result in the E motor skipping - for example if the filament is soft enough that it gets ground by the extruder wheel.

For those who are getting stuck filament out of the blue, we still haven't been able to verify any situation where this warning popped up and the extruder did not actually skip - so I would guess that your extruder did indeed skip steps. This can happen even without the filament getting stuck - it can be as partial clog, or the hotend simply not being able to fully keep up.

bryn51, 2025-01-04:
Thanks for the explanation. So I'm reading that the detection is based on e-motor being asked to rotate and it not rotating.
Eg. An indirect way of detecting the "stuck" condition.

Whereas "real" stuck filament occurs when the emotor rotates but the filament does not move and some proportion of the time it will occur when filament gear  slips against the filament eg "grinding".
Most often this will happen when nozzle tube or nozzle orifice is fully or partially blocked. But can also occur when filament input movement from the roll is impeded eg due to a knot on the roll or a kink.

The user is left to diagnose which of these several conditions occurred. But the detection is only for a subset of possible issues.
So, whats needed is a filament movement detector that is read in conjunction with filament feed instructions eg a feedback system.

I note that the Core One printer spec indicates an additional filament detector is fitted. Nothing more is said about it so it's a mystery to customers. Could it be the additional sensor is an attempt to improve the above situation ?
If so could the mk4s have an upgrade to include the additional sensor ? 
Surely improvement is needed, since full or partial clogs are not unknown on mk4s, abd surely other feeding issues can also occur.
The current detected condition of skipping e-motor actually quite rare, I think.(although it happened to me, obviously)

bryn51, 2025-01-04:
I also note that there is a world of difference between a "filament presence" sensor, and a "filament movement" sensor, and one needs to be careful with the use of terminology (and I think prusa do not use such language precision)

CZDanol, 2025-01-05:
Filament sensor on the C1 is just a standard side filament sensor that works in the same way as on the XL - because of the coreXY mechanics and enclosedness of the machine, the C1 has a long-ish PTFE tube from the extruder to the side compartment where the filament is mounted. If there would be no side filament sensor, we would be able to detect filament runouts only when the filament runs out at the extruder, which would mean that you would have to remove the top lid and extract the remaining small piece of filament from inside the PTFE tube. With the side FS, we detect it earlier and the user can pull the (albeit longer) remaining filament from the side compartment directly.

Also, it detects sooner when the user is trying to insert a filament and moves the head to an optimal position for inserting (smaller bending radius of the PTFE tube, so the insertion is easier).

Prusa currently does not need this language precision, because we don't have any filament movement sensors in our released products - so all "filament sensors" are just the presence ones.

bryn51, 2025-01-05:
Thanks for clarifying on both items.

Yeah I agree that a filament movement detector would be like the holy grail of sensors, but likely it would be somewhat more costly than a presence sensor (which can be made in several ways, but generally inexpensive).
But if it could be made economically, it would find a great home in (say) the MMU3, for example as well as in the extruder.

I can, in my minds eye, think of a general design for such a sensor, but the trick would be to make it compact as well as inexpensive.

> Filament sensor on the C1 is just a standard side filament sensor that works in the same way as on the XL - because of the coreXY mechanics and enclosedness of the machine, the C1 has a long-ish [...]

CZDanol, 2025-01-05:
I would say that for the "holy grail", you would need both presence and motion detection - you want to know whether there is a filament present even before you try moving it.

bryn51, 2025-01-05:
yes of course, but as aforesaid, today we have only the one, "presence". Plus we have the stuck filament detection schema, there "we ask the motor to rotate, but it does not".  So, I'm suggesting, when we have "filament movement detector" we widen the perspective, and say "we ask the motor to move the filament, but it does not (so it must be stuck)". And then the feature meets user expectations, and by the way we solve a whole bunch of issues experienced by users in this otherwise really great printer.  
Onward and upward ?

JohnOCFII, 2025-01-05:
> I have no idea if this thing works, or how much it would cost in production quantity, but what I'm suggesting isn't a pipe dream.

There are a variety of "filament movement" sensors out there.  I've built and use this one - and have used it with both RRF and Klipper.  The Marlin-based Prusa firmware would need to be configured to support it. Parts cost is minimal, but it does require some tuning to avoid false positive alerts.   If I were Prusa Research, my calculations would have to include, "how much time will my support people spend supporting this option, and is it worth the added time?" 

https://www.printables.com/model/629380-filament-progression-and-runout-sensor-optical-end

m10d, 2025-01-27:
> There are a variety of "filament movement" sensors out there. I've built and use this one - and have used it with both RRF and Klipper. The Marlin-based Prusa firmware would need to be configured to [...]

It has already been attempted and discarded - [mk3 used a optical sensor](https://help.prusa3d.com/article/filament-sensor-mk3-non-s_2125) similar to (probably derived from) a mouse.  It proved too problematic to support in all circumstances, and was discarded (literally replaced with a non-movement-sensing approach with mk3s.

So, if you want to discuss movement sensing that's fine, and while I'm sure theoretically it could be done, but Prusa has already looked into it *extensively* as they shipped it at one point in quantity ... lets keep that discussion to a "feature request" because that's what it is, and since it requires hardware it's *totally* irrelevant to this **BUG** request.

m10d, 2025-01-27:
I came here because I see the tangle-type stuck filament on my xl5t, with no detection.

> Hey, the "stuck filament" feature is named a bit inappropriately - what it actually does is that it detects the E motor skipping. The filament can however get stuck in a way that does not result in [...]

Is that true? While prusa does sometimes fail to update the support pages, at least right now they seem to[ directly contradict what you're saying](https://help.prusa3d.com/article/stuck-filament-detection-26101-mk4s-13101-mk4-21101-mk3-9-17108-xl_690089) about the e-motor (or rather the drivers performing skipped-step detection). Emphasis mine:

> It is important to note that detecting stuck filament is possible only inside the Nextruder, as pressure builds up from the nozzle to the upper part of the Nextruder. However, if your filament is tangled on the spool, there will be no internal pressure, just outside tension, which **_the loadcell won't be able to recognize_**.

I can't find a link, but use of the loadcell was touted in the nextruder's release as well.

What's strange to me is that Prusa state it can't detect the tangle-based.  This would seem to be the _easiest_ case.  Forward force on the nozzle just remains near zero, it does not vary with extrusion.   Due to the nature of partial clogs, irregular filament, filament with high solids content, it would seem MOST difficult to set a good threshold or heuristic to fire stuck filament based on too HIGH of force.

Alas, on **XL5t fw 6.1.3** I experience this failure, which is the same as the tangle case - **nextruder gears grind the plastic**, nothing extrudes, but the print continues merrily along.

bryn51, 2025-01-27:
Yes.  Its been discussed at length. Its really a printer design issue rather than firmware.  The printer lacks a filament movement sensor, so it is unable to tell if the filament moves when the e-motor turns, or not.

The firmware issue is really in the labelling of the existing stuck filament error detection. It says "stuck filament detection" (to turn it off and on), but really its something else, like "e-motor skipping - filament not being driven" to avoid misunderstanding by users.

There are filament movement detectors out in the wild, but fitting one onto the printer would be another thing all together, and then it would need to be integrated into the firmware.

m10d, 2025-01-27:
> Yes. Its been discussed at length. Its really a printer design issue rather than firmware. The printer lacks a filament movement sensor, so it is unable to tell if the filament moves when the e-motor turns, or not.
> 
> The firmware issue is really in the labeling of the existing stuck filament error detection. It says "stuck filament detection" (to turn it off and on), but really its something else, like "e-motor skipping - filament not being driven" to avoid misunderstanding by users.
> 
> There are filament movement detectors out in the wild, but fitting one onto the printer would be another thing all together, and then it would need to be integrated into the firmware.

Yes to what?  The question I asked was about whether filament stuck detection on the nextruder is using the loadcell, as is stated in prusa's help pages & initial advertising, but was contradicted by the commenter I quoted.

I'm replying here for ^, but since you continue to go down the hole of optical movement (which would be great, but this is a BUG report, and you're very clearly clouding the water by spending space on a _feature request_ (and a hardware one, at that, in the fw repo) I'll ask as well:   What_ printers are shipping with movement sensors a-la the mk3?   I'm not aware of any, at least limiting the discussion of course to well-supported devices in the hobby space.  I know of several that use optical _presence sensing_, but this is functionally equivalent to mechanical movement + hall effect, each of which have their own drawbacks.  BL X&P series use a system pretty similar to the nextruder (the BL filament physically displaces a mechanical arm, the nextruder displaces a ball).  BL's is definitely not motion according to their troubleshooting page.  Not sure what AMS uses, I read conflicting statements - but that's not terribly relevant as the print can't fail to pause if the _extruder_ is sensing robustly (sure it might be a PITA if a sensor in the MMU or AMS fails - but it's not in and of itself going to ruin a long print)

bryn51, 2025-01-27:
> Yes to what? The question I asked was about whether filament stuck detection on the nextruder is using the loadcell, as is stated in prusa's help pages & initial advertising, but was contradicted by [...]

I cannot see why you are adopting that tone. There is no argument between what you and I are saying.  It originally seemed  to me and others like it was a bug as the filament was evidently stuck, but not being detected. So now we understand it a little more seems more like its an enhancement thats required.  THis sort of thing happens, and surely flexibility is whats needed.  But to date we see no comment by Prusa on this thread, so it all seems futile.  This is not the first time for that.

bryn51, 2025-03-31:
thanks for ignoring the issue.
it would be far more helpful if you could say that it has been referred to the hardware design team for consideration.

rob-miller, 2025-03-31:
I was having a variation on this problem last week, admittedly down to old filament but repeated several times.  

The filament would break between the Nextruder drive gear and the top of nozzle tube.  I surmise that the filament from the drive gear was then out of alignment with the broken piece so could not push it through; at this point the load cell should have detected it as stuck, but did not.  What did seem to work was the retraction moves, as what I would find when I stopped it 'printing in air' was the incoming filament fully removed from the drive gear area.

I think that a useful thing might be a re-design of the idler lever to put more pressure on the drive gear (to trigger the load cell) when the filament is not present.  There are points/ridges on the current design that are higher than the idler wheel, perhaps these could be emphasized in some way (even to the point of fixed metal teeth?) so that the drag on the drive gear goes up significantly if the filament is not present.

This might cause a problem loading filament, but from inspection the filament is touching the idler wheel at the point of loading so might lift it up in time.

While this would not save a print as effectively as pausing as soon as the filament is stuck initially, detecting the situation at all would be better than the current situation.

bryn51, 2025-06-02:
Well today I was researching Bambu's H2D printer, and found they have implemented a filament movement sensor as part of their AMS2 Pro. 
So, now I must ask why Prusa are prepared to close this issue, when its clear competitor has identified it as an issue and found a way to fix it. Fixing of design flaws of this nature should not be popularity contest.

closed 2025-08-12 by github-actions[bot]

rob-miller, 2025-11-02:
> My prusa xl 5-tool is basically useless. It's constantly getting stuck
> filament issues with PETG. I fixed it on tool head 1, I think, by tweaking
> some filament settings. But tool head 2 is completely useless. The layers
> are awful. It can't go 4 moves without a atuck filament sensor going off.
>
> What do I do??
>
> I'm on firmware 6.2.6+8948. Prusa 5-tool circa 2023 with the enclosure.
> All tools have (lightly used at this point) 0.4 HF Obxidian nozzles.

That’s interesting. I haven’t tried my HF obxidian nozzles yet, but given
the design I’m not surprised that they would have more back pressure and
thus trigger this sensor.

I would start with raising the extruder temperature to the top of the PETG
range. If that doesn’t work we probably will need more range in the
sensitivity here (firmware).

m10d, 2025-11-05:
> I'm not into playing word games. In fact there is no syntactical difference in practical terms between a "bug" and an "enhancement". There are only ever "code changes" anyway.
> 
> From a real world ....

Look bud - in the real world, we're interacting with a company & open source developers with their own goals priorities and predilections.  Sometimes it may not matter, but sometimes it will. And since _you_ are not _them_, take a step back and just be accurate with the terms.   While as a fellow user I also (strongly) want the product to cover _all_ of what we would colloquially call "stuck" - the help page is _very_ clear what they mean by "stuck" at this time.   So if it doesn't do what it says it does ... that's a bug.  If it doesn't do something you want it to do, that's a feature.   If you need to get worked up about be verbiage, start calling your stuff `user stories` as that seems closer to what you're trying to grasp for.

More importantly, you're arguing some verbiage in my post - which is actually clearly about making the issues captured unambiguously.  There are two.  They overlap.  This confusion can help no one, not users, duplicate bug submissions, or the dev teams themselves unless they indicate they'd prefer it broken down differently (but their way would certainly be _unambiguous_ which is all i'm after)

Nor does just lumping anything under the sun that strikes you as "stuck" into one issue.  There is a reality of how it's implemented strongly influencing what's fixed, how, and when ...

bryn51, 2025-11-05:
As I read it, you are just arguing for the sake of arguing. perhaps you just like doing that. I have better things to do. I suggest you just have a Bex and a lie down, cool off.

bryn51, 2026-05-18:
Its clear other folks have this same issue.  In fact it occurs on all Prusa Printers I currently own (No longer owning a MK4, but do own a CoreOne and a Prusa XL, and it happens to them too. )
I also own a recently acquired Bambu H2D.  And I noticed, the printer detected when it was having trouble feeding Brand X PETG-CF  thru a 0.4 mm nozzle, it paused the print immediately and alerted me. I was able to purge the nozzle and carry on the print, its not lost. If Bambu can do this, why not Prusa. ??  I pay more for the privilege of owning a Prusa, why is it not as good as a printer costing 60% of the price :? And I just do not accept that you just let the issue go stale without taking some action about this widely experienced flaw.