prusa3d/Prusa-Firmware-Buddy #4088: [BUG] Prusa XL not initializing bed/board/etc temp sensors

goodridm, 2024-07-17:
### Printer model

XL 5 Tool

### Firmware version

6.0.3

### Upgrades and modifications

None

### Printing from...

USB Drive

### Describe the bug

Prusa XL 5 Tool not showing Board, Bed, or Dwarf MCU temperature values on display during a print nor when idle.

[image]
[image]
[image]

danopernis, 2024-07-18:
Hi @goodridm thanks for reporting. I wasn't able to reproduce your issue. Is there a specific series of steps you need to take to get to this state? Did this work in different version of firmware but stopped working after an update? Since nobody else is reporting this, I suspect this is hardware issue.

goodridm, 2024-07-18:
This did not occur in previous firmware versions, or I at least didn't notice the temps not showing. It seems to be intermittent, a power cycle this morning restored the temps showing. Not sure if it occurred because of the firmware flash to 6.0.3 and it needed a power cycle after to clear. I will continue to monitor and reply if it stops reporting again.

Pyrunner, 2024-07-19:
I have an XL dual and after this firmware install yesterday printer does not pass bed heater test and reports mismatch of filament when not true. Online chat offers no resolve

danopernis, 2024-07-22:
@Pyrunner your problem seems unrelated to this issue which is about sensor values being displayed in the sensor menu. This is not a forum and we would like to keep comments on the issues as relevant as possible. I suggest creating separate bug report with as many details as possible. Thank you for understanding.

MalikDrako, 2024-07-30:
I'm seeing this as well on a Mk3.5. It seems random if/which sensors are not initialized.
I did just check while printing from Octoprint and several were not initialized

danopernis, 2024-08-07:
Looking at the code, `SensorDataBuffer` seems overly complicated and prone to failures. It warrants complete rewrite. I have created internal ticket for that.

psimolin, 2024-08-29:
I am experiencing the same issue for the first time today. Prusa XL 5TH, latest firmware.

CZDanol, 2024-10-31:
The "not initialized" sensor issue should be fixed in 6.2.0-alpha. Please let us know if the issue persists.

Prusa-Support, 2025-02-13:
Thank you for your participation in this activity on GitHub.
The issue can be closed as solved now.

Michele Moramarco
Prusa Research

closed 2025-02-13 by Prusa-Support