prusa3d/Prusa-Firmware-Buddy #4091: [ENHANCEMENT] show selection of steel sheet in menu

cnlson, 2024-07-18:
### Printer model

mk3.5

### Describe the enhancement

when you make a selection in the steel sheet menu, under hardware, The selection is not highlighted or shown in a different color so it is impossible to determine which sheet is selected. 

### Expected functionality

in the mk3s+ the sheet used was shown directly in settings. In the mk3.5 with a color screen the sheet selection should be either highlighted or displayed in a different color. also custom sheets should be renamable.

also, when you select a sheet, rather than having to back out of the menu you should be returned to the sheet selection page with the now selected page highlighted or displayed in a different color to indicate the selection.

Printer606, 2024-07-18:
I would also want this to happen.

cnlson, 2024-07-18:
> I would also want this to happen.

well i guess we are in luck! if you save more than one sheet it is displayed under control/set ready 
[image]

cnlson, 2024-07-18:
i just had to save a couple of sheets before it displayed this field under the control menu.

closed 2024-07-18 by cnlson