prusa3d/Prusa-Firmware-Buddy #4098: Starting a print when Extruder is far right, results in attempt to drive extruder down into the bed with force.

jsrumsey, 2024-07-20:
### Printer model

MK3.5 with MMU3

### Firmware version

6.0.4

### Upgrades and modifications

None

### Printing from...

USB Drive

### Describe the bug

Starting a print when Extruder is far right, results in attempt to drive extruder down into the bed with force - perhaps Pinda is over the right edge and can't detect the bed?  It forces the bed down, slightly bending it.  Instead it should home back to the XY home position first.

### How to reproduce

Move the X to full right and then start a print.  It may also be a load filament command.

murk-sy, 2024-07-25:
Related issues since this was a problem on the Mini as well, but on the left side instead:
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/810
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1268
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1182

tl;dr: Disabling steppers or losing known X position in any way (possibly Y as well? not sure if an issue on MK3.5) should require homing X before any automated (especially downwards) Z moves. Seems this was missed for 3.5, but the Mini specific code can probably be reused

jsrumsey, 2024-07-28:
I believe this has been resolved by cleaning out old printer configs and creating a fresh config.

jsrumsey, 2024-08-05:
It failed again, after I stopped a print via Prusa Connect, before it had starting printing, but the head was moving down to the bed. It was in full X to the right position and kept going down below the bed bending the sheet down, with Pinda sensor hanging off to the right of the bed. The printer LCD interface was unresponsive, and I had to hit the reset button. I then had to recalibrate the Z.

ripnyt-ripnyt, 2024-08-27:
this bug has occured for me more than a few times. very irritating, and now I don't know if my bed/y-carriage is square and flat any longer.

jsrumsey, 2024-08-27:
After I had a chat with Prusa on another (related?) issue, they had me do a factory reset, and then the next day they came out with another firmware update to 6.1 and so far I haven’t seen this problem or the other two.  It was acting like the firmware thought my MK3.5 was a different model, like an MK4.  I’ll give it a little more time and then close it if no further reoccurrence.

closed 2024-08-27 by jsrumsey