prusa3d/Prusa-Firmware-Buddy #4099: [ENHANCEMENT] Allow upload of new firmware via Prusa Link

RichiH, 2024-07-21:
### Printer model

XL

### Describe the enhancement

Currently I can only upload, and delete, .gcode and .bgcode files via Prusa Link.

### Expected functionality

I'd like to be able to upload .bbf as well

RichiH, 2024-07-21:
@pstrobl96 alongside https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3428 maybe there's a chance to bump this internally? Seems ~obvious.

m-cas, 2024-07-25:
I like the idea of not having to remove the memstick anymore to add new firmware so that a fw upgrade has only one trip to the printer for the restart. I guess the next step would be a button in the GUI to restart the printer after having uploaded the new firmware without having to confirm the upgrade on the printer display?

Asuanin, 2025-01-04:
+1
Printer model: MK4

TheRealBoss, 2025-01-29:
+1 for Mini+

Whitelist .bff

samdurnxn, 2025-08-01:
It will be a great enhancement. +1 in the next version of the firmware if possible ^^

closed 2026-10-05 by github-actions[bot]