prusa3d/Prusa-Firmware-Buddy #4100: [BUG]  MK4 Extruder do fake purge in the air, as print is finished (difficult to describe)

Mercur13, 2024-07-21:
### Printer model

MK4+MMU3

### Firmware version

6.0.3

### Upgrades and modifications

None

### Printing from...

Octoprint

### Describe the bug

I printed a 2-color flat object together with a higher part in the base color. (see photo)
The cleaning tower is also printed correctly up to the height of the last layer in the additional color. So everything is OK.
The higher object is then printed in the base color. The cleaning tower is not built up any further.
I noticed that after the last layer the extruder moves over the position of the cleaning tower several times, as if it wanted to clean itself again (but in the air). However, no filament is extruded.
[image]

### How to reproduce

as described. 3mf-File attached
[4-secondary-2_color-print-group.zip](https://github.com/user-attachments/files/16323874/4-secondary-2_color-print-group.zip)

ojrik, 2024-07-21:
This error is already long-standing in SLICER. It is not an error in the Firmware.

ojrik, 2024-07-22:
It's hard to sort through 4100 improvements and bugs. Example this is new #13070. https://github.com/prusa3d/PrusaSlicer/issues/13070

closed 2024-09-30 by github-actions[bot]