prusa3d/Prusa-Firmware-Buddy #4101: [ENHANCEMENT] XL Filament Auto Load doesn't home extruder before loading the filament

rcmaniac25, 2024-07-22:
### Printer model

XL Single Tool (Firmware 6.0.3)

### Describe the enhancement

While doing a set of prints, I had done a large print. To remove it, I pushed the extruder all the way to the back of the printer. The same spot that a multi-tool XL would park the unused extruders, which is outside the print volume.

I then inserted filament to load it. The autoload filament feature kicked in, drew the filament in a little, and started pre-heating. I realized that if it doesn't move the extruder, it will autoload the filament behind the printer volume, dropping it behind my workbench to a location I can't reach easily. I manually moved the extruder forwards right before it started purging filament.

After testing filament unload and (manual) filament load, done via the menu options, I realized both of those options home the extruder before loading or unloading filament. I felt that the XL should do the same when using autoload.

This is a copy of #3417 but I couldn't reopen the ticket

### Expected functionality

When filament is pushed into the extruder to auto-load, it should home itself before loading. The same way manual filament loading and unloading does.

Prusa-Support, 2025-09-19:
With FW 6.4.0-alpha, auto-loading has been improved in several ways.

As a side note...
> ... In this case, the toolhead moves to an optimal position for feeding filament through the PTFE tube, and the extruder begins spinning to assist insertion. ...

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.4.0-alpha#:~:text=toolhead-,moves

This leads me to believe that filament loading and purging outside of the print area should no longer happen during the filament auto-load process.

I wonder if the issue can actually be considered solved now.
Possibly, please test FW 6.4.0-alpha and provide feedback.

Michele Moramarco
Prusa Research

rcmaniac25, 2025-10-15:
@Prusa-Support Sorry for the delay, I tested and the issue is still there. If I simply insert the filament to the toolhead, it triggers autoloading. The bed lowers a little, but the toolhead does not home itself. So per my description of the situation, it extruded behind the print area (and behind the printer).

rcmaniac25, 2025-12-15:
I have the release version of 6.4.0 installed and the same situation applies, so was not fixed.