prusa3d/Prusa-Firmware-Buddy #4105: [ENHANCEMENT] Screen Saver

ALD-Models, 2024-07-28:
### Printer model

Mini, Mk4, XL

### Describe the enhancement

Would it be possible to preserve the life of the display so after x amount of time of inactivity the screen will dim and show a bouncing Prusa logo to preserve the pixels?

This would especially be a good feature when you have a printer scheduled to print for the following morning you don't have the display on all night!

murk-sy, 2024-07-30:
Duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/660

It was closed as stale but I'll make a comment there as well to perhaps have it reopened, since there's a lot of information there already. I also have a suggestion as well.
I don't think burn-in is a particularly big concern though.

ClaGre, 2024-08-01:
Otherwise a simple clock

closed 2024-08-06 by danopernis