prusa3d/Prusa-Firmware-Buddy #4107: [BUG] Changing filament on Prusa XL 5TH on Tool4 also preheats Tool1

Investegate, 2024-07-29:
### Printer model

Prusa XL 5TH

### Firmware version

6.0.4

### Printing from...

USB

### Describe the bug

For example, if I change the filament on tool 4, tool 1 is always heated to 170c even though I haven't changed anything... I then have to deactivate it manually.

MalikDrako, 2024-07-30:
I'm seeing this too, but I think it's a bit more general than this.
After changing/loading/unloading the filament on any tool other than T1, T1 is set to the standby temperature (170C), while the active tool remains at the change temperature.

Expected behavior: the active tool is set to the standby temperature, all other tools remain unchanged

morphias2004, 2024-08-16:
This has already been raised in #3552 

Please add comments and upvotes there.

closed 2024-09-17 by danopernis

morphias2004, 2024-09-19:
This absolutely can be reproduced. Why is this not planned to be fixed? It's a pretty simple code change to NOT turn on E1 when performing loads/unloads on other tool heads.