prusa3d/Prusa-Firmware-Buddy #4109: [ENHANCEMENT] Post print screen, replace "Tune" with "Set ready".

planeturban, 2024-07-30:
### Printer model

Mini+

### Describe the enhancement

After a print from queue is done one have to press *gasp* three times, plus jog the wheel some, to print next in queue.

Suggestion: After a print is done, replace "Tune" with "Set ready", but still select "Reprint" (or whatever it says). Makes for a better and quicker experience.

### Expected functionality

1. Print from queue.
2. Wait
3. Remove print
4. Jog wheel once to the left and press.
5. Profit.

BugsBunny1403, 2024-08-08:
Nice idea

danopernis, 2024-08-08:
This looks like a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4069

closed 2024-08-08 by danopernis

planeturban, 2024-08-08:
> This looks like a duplicate of #4069

Surely does, great minds…