prusa3d/Prusa-Firmware-Buddy #4111: [BUG]  6.0.4 MK4 Y Axis Calibration Fail

mhornsby, 2024-07-30:
### Printer model

MK4

### Firmware version

6.04

### Upgrades and modifications

none

### Printing from...

USB Drive

### Describe the bug

Yesterday I completed a 40 hour print, then upgraded from 6.03 to 6.04. When to print this morning and the printed failed Y calibration. Run the Calibration test and it failed at the start of the Y calibration homing to the front of the printer. Downgraded to 6.03 and the Y Calibration worked. Factory Assembled MK4. Did a belt test 88/89 Hz.

### How to reproduce

Upgrade to 6.04 from 6.03

### Expected behavior

Y Axis Calibration Pass

mhornsby, 2024-08-01:
Ran a few days later and it worked so you can close the issue

closed 2024-08-01 by mhornsby

graelo, 2024-08-14:
Hi, I did encounter this too, but TBH the furniture the printer was on at that time was a bit wobbly. Could it also have been your case?