prusa3d/Prusa-Firmware-Buddy #4113: Filament Loading Errors Screen [BUG]

mj8w, 2024-08-03:
### Printer model

MK3.5 + MMU3

### Firmware version

6.03

### Printing from...

PrusaConnect and directly from front panel

### Describe the bug

I get the message "Printer has detected multiple consecutive filament loading errors. We recommend checking Nextruder main-plate."

1.) MK3.5 does not have a main-plate
2) How do I turn this message off - how will I know if I fixed it?

### How to reproduce

I did have some problems with loading, my extruder PTFE tube was clogged. I replaced it, but I don't know if it is still having problems or , because I can't reset the status.

### Expected behavior

Error screen should be custom for MK3.5
It should tell you how to reset the status too.

ClaGre, 2024-08-04:
Same 4 me and I've to reset the printer.
My HE is Revo, when I want to print Polyterra PLA the knurled roller on MMU3 sometimes grinding the filament when the idler try to pull the filament and MMU can't extract the filament from the nozzle.

danopernis, 2024-08-06:
Thanks for reporting. This is a valid issue and I have created an internal ticket to fix it. There is a menu item in "Hardware" menu to reset the status, but it is disabled for MK3.5 which is clearly a bug. References to nextruder are cosmetic, but should also be fixed to prevent confusion.

ClaGre, 2024-08-06:
Here photo of my information screen.
[image]

rainer1146, 2024-09-04:
The same with the MK4. Even if I print directly into the Nexdruder without the MMU, the error message appears.
Even if I only unload the filament.

closed 2025-12-09 by github-actions[bot]