prusa3d/Prusa-Firmware-Buddy #4118: MK3.5 Power Panic sets hotend temp to 170 instead of active filament temperature [BUG]

phillipscheng, 2024-08-04:
### Printer model

MK3.5

### Firmware version

6.0.4+14924

### Printing from...

USB / PrusaLink

### Describe the bug

On power panic reset, the printer returns to 170C instead of the active filament temperature.  I suspect that this is due to the partial homing on return from power loss.

### How to reproduce

Power panic seems to trigger the issue about half of the time.

### Expected behavior

The printer should return to the current set temperature for the active filament.

phillipscheng, 2024-08-21:
Possibly the same bug as https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4096

phillipscheng, 2024-10-21:
This issue still occurs.

CZDanol, 2024-10-21:
@phillipscheng have you tested this on 6.2.0-alpha2?

closed 2024-12-30 by github-actions[bot]