prusa3d/Prusa-Firmware-Buddy #4125: [FEATURE REQUEST] Project thumbnail page when printing for the MINI/+

Printer606, 2024-08-06:
### Printer model

MINI/+

### Describe the feature

When printing, it will show the image of the project being printed

### Expected functionality

Quite obvious.

BugsBunny1403, 2024-08-08:
Nice idea.

Printer606, 2024-08-16:
> I think this would also potentially be a good place to show the "status light bar", but on the screen itself. Or, alternatively, change the progress bar to use the lightbar colors. Right now, at a glance from afar or a darkened print room it may be somewhat difficult to see the status. I doubt Mini is often used in farms, but still.
> 
> ![image](https://private-user-images.githubusercontent.com/1211672/358538525-405fceba-ecf5-40c3-8298-d0977a23373e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM4MjIzNTIsIm5iZiI6MTcyMzgyMjA1MiwicGF0aCI6Ii8xMjExNjcyLzM1ODUzODUyNS00MDVmY2ViYS1lY2Y1LTQwYzMtODI5OC1kMDk3N2EyMzM3M2UucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgxNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MTZUMTUyNzMyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjMyMTllY2JlNDZkYzkwMTA0NmU5YjkxYzc2NWIzNzhmNjhkNzMxNDEwYjQxZGVhZjBkM2Q5M2JiZDI4ZWFhNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.x4paV9UGMWIRpL4J3fzm-hK4sWjfJf884wK1_NE9cA0)
> 
> Render quality mspaint mockups: Added-on lower bit of the screen as indicator ![image](https://private-user-images.githubusercontent.com/1211672/358538121-e5cc5d27-eb6c-4efc-8ba6-67e75473a1e8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM4MjIzNTIsIm5iZiI6MTcyMzgyMjA1MiwicGF0aCI6Ii8xMjExNjcyLzM1ODUzODEyMS1lNWNjNWQyNy1lYjZjLTRlZmMtOGJhNi02N2U3NTQ3M2ExZTgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgxNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MTZUMTUyNzMyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YWYxZGE4YTAwNDIxZTUxNjgzNmVhMWY4MTEyOTZhYTdjZDJhMWVhYzQwNDBiZjVkMDMyNzY3YmY4ZWRiMzZmNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.9FClHygiZSLhk9x2iAPMjY7a27sIdJCFR3_VCIU47jw)
> 
> Colored progress bar ![image](https://private-user-images.githubusercontent.com/1211672/358539863-43b75ac1-722f-4d6e-975f-8593c18fe5af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM4MjIzNTIsIm5iZiI6MTcyMzgyMjA1MiwicGF0aCI6Ii8xMjExNjcyLzM1ODUzOTg2My00M2I3NWFjMS03MjJmLTRkNmUtOTc1Zi04NTkzYzE4ZmU1YWYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgxNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MTZUMTUyNzMyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NmM1YjFmYWYzMDI4MTUyNDM5M2Q3NDFkZjc1YTdjNDFhYmMzMGMzZGIyNWI1ZTJiZGYzYjAzNDZmMTZhYTlmOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.R52UWzwvMo_kxarlNPpLJDyrr0fjGPQDASOHrLSyL90)
> 
> Caveats:
> 
> * Product department would probably hate this because it's somewhat inconsistent with the color scheme
> * Status light bar may be a future addition to newer hardware if and when the Mini gets an upgrade path

That is a really good idea!

closed 2025-11-13 by github-actions[bot]