prusa3d/Prusa-Firmware-Buddy #4126: [FEATURE REQUEST] Add customizable buttons to home screen

me25542, 2024-08-07:
### Printer model

MK4, XL

### Describe the feature

The main benefit of adding customizable home screen buttons, as I see it, is time saving. The ability to, in effect, select your most used item(s) from either the control or filament menus would save time navigating through said menus. This might be especially useful for prusa connect users, like in print farms (it might improve efficiency in _your_ print farm, prusa).

### Expected functionality

Similar to the customization of the footer, in settings users would be able to select items (controls or simple on/off settings) to add to the home screen. These would either take the place of the preheat button, (the preheat button would have its menu moved inside the control menu as a sub-menu), or as two extra buttons on the home screen. (see attached photo)
Items that could be added to the home screen might include:
set ready (prusa connect), move axis, temperature, auto home, disable motors, Cold pull, Calibration and tests, load filament, unload filament, change filament, purge filament, preheat, stealth mode on/off, filament sensor on/off, and stuck filament detection on/off. (Basically anything from the control, filament, or settings menus)

(Sorry for the poor example photo quality, i made it in ms paint)
[image]

BugsBunny1403, 2024-08-08:
I need this. Good idea.

ildryan, 2024-08-14:
If I may add: a shortcut to nozzle diameter change.

Relatively speaking, it feels longer to navigate to and change the nozzle diameter on the screen than changing the nozzle itself on the printer...

rob-miller, 2025-04-22:
I would use customizable button to park current tool

me25542, 2026-09-14:
This would still be nice to have.