prusa3d/Prusa-Firmware-Buddy #4127: [BUG]

Emusp, 2024-08-07:
### Printer model

Mini+

### Firmware version

6.0.4

### Upgrades and modifications

None

### Printing from...

USB Drive

### Describe the bug

If you start a print directly after making a filament change, the extrude will crash into the bed.
I does not auto home before starting print. 

### How to reproduce

Do a filament change, then start a new print. 

### Expected behavior

Printer should auto home before staring print.

jsrumsey, 2024-08-07:
This sounds very similar to the issue I've seen -- #4098
Jeff

Emusp, 2024-08-07:
> This sounds very similar to the issue I've seen -- #4098 Jeff

Could be the same issue.
It's crashing in the filament change parked position all the way to the left.

Will try to capture a video.

Tomcus, 2024-08-08:
Hi @Emusp, thanks for reporting the issue, but I am unable to reproduce it. Could you provide a video or more detailed steps how to reproduce it? Also it might be an HW issue, can you pleease check if the Pinda is working? On MINI you should see a red light comming from it when it doesn't detect any metal, you can put screwdriver under it and the light should turn off.

Emusp, 2024-08-08:
> Hi @Emusp, thanks for reporting the issue, but I am unable to reproduce it. Could you provide a video or more detailed steps how to reproduce it? Also it might be an HW issue, can you pleease check [...]

Video of the issue: https://youtu.be/bMu9M9FGY10

Pinda is working, and the printer is working fine otherwise. I just need to remember to manually auto home before I start a print after filament change.

Tomcus, 2024-08-08:
My gcode, that I have printed (and had no problems) did have a gcode to home before printing. Can you check if the gcode file you are trying to print contains `G28` gcode? Because it doesn't look to me that the printer tried to home in X and Y, or maybe the printer thought that the printer is already homed on X and Y. If you have a binary gcode you can convert it in prusa slicer (File > Convert > ...).

Emusp, 2024-08-08:
G28 is present in the converted gcode.
So it should home, but it skips the homing and goes straight to G29.

But when I start the print again after the crash, it will auto home fine and run the print normal. 
It's only happening after a filament change.

Emusp, 2024-08-08:
I just found out that it's only crashing/not homing when you do a filament change as the first thing after the machine as been booted.
Turn on the machine -> Change filament -> Press STOP -> Start a new print -> crash. 
Turn on the machine -> Start a print -> Cancel it -> Change filament -> Press STOP -> Start a new print -> Auto Home fine

Tomcus, 2024-08-08:
Ok, I will try that.

Tomcus, 2024-08-08:
It worked for me fine. Have you experienced this issue before `6.0.4`?

Emusp, 2024-08-08:
Yes, in  6.x.x
Not sure in 5.x.x

Tomcus, 2024-08-08:
@Emusp Please try to contact Prusa Support, since I am unable to reproduce the problem. It might be a known problem for them, and you will need just to tighten something. I will keep this issue open, maybe someone will have the same issue with different reproducer.

Emusp, 2024-08-08:
I tried a Factory Reset and FW re-flash with 6.0.4 + calibrations. 

The printer seems boot way faster now. But unfortunately the crash behavior presist.

murk-sy, 2024-08-16:
Could not really reproduce it with instructions in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4127#issuecomment-2275600696
I did manage to discover an issue that is likely related, see mention.

What I tried was to trigger filament change and when the print head is in position, stopping it and then running the below gcode (default start gcode for mini but without any heating or printing). Interrupting while it's moving to the left didn't reproduce it either.

[2024-08-16 issue 4127 start gcode only.zip](https://github.com/user-attachments/files/16636227/2024-08-16.issue.4127.start.gcode.only.zip)
```gcode
M862.3 P "MINI" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M862.5 P2 ; g-code level check
M862.6 P"Input shaper" ; FW feature check
M115 U6.0.4+14924
G90 ; use absolute coordinates
M83 ; extruder relative mode
M569 S1 X Y ; set stealthchop for X Y
M204 T1250 ; set travel acceleration
G28 ; home all without mesh bed level
G29 ; mesh bed leveling 
```

Can you try reproducing it with the above print file and check all the steps again, also specifically when you press stop?

There's been similar issues but I don't any of them were during a print. If G28 is in the gcode it will be run, regardless if it needs to or not.