prusa3d/Prusa-Firmware-Buddy #4128: [BUG] Print resume after power loss continues at lower Z level

honza03210, 2024-08-08:
### Printer model

MK3.5

### Firmware version

6.0.4+14924

### Upgrades and modifications

E3D Revo six hotend

### Printing from...

Prusa Connect

### Describe the bug

Printing with 0.25 nozzle, 0.15mm layer height. 
During the first layer, I powered off the printer for a few seconds to simulate power loss as I wanted to check that it works correctly (a risk of strong thunderstorms today during a long print). Upon recovery, the printer resumed at lower Z causing the nozzle to make rather deep scratch into the PEI sheet.

### How to reproduce

I didn't try to reproduce, don't have unlimited printing sheets :)

### Files

I was printing from .bgcode, the gcode in the zip was generated with all the same settings as the original bgcode.
Power panic kicked in in the middle of the second keychain.
[gcodes.zip](https://github.com/user-attachments/files/16538769/gcodes.zip)

danopernis, 2024-08-14:
Thanks for reporting @honza03210 this appears to be reproducible. I have created an internal ticket for further investigation.

3d-gussner, 2025-09-25:
@honza03210 Can you please verify if this issue has been solved with the newest firmware? If so please consider to close it.