prusa3d/Prusa-Firmware-Buddy #4131: [BUG] MK4 cannot read files copied from google drive for desktop

0ba-pia, 2024-08-10:
### Printer model

MK4

### Firmware version

6.0.3

### Printing from...

USB Stick

### Describe the bug

.bgcode files that are stored with google drive for desktop (windows version) and copied to the prusa USB stick won't be listed by the printer.

0ba-pia, 2024-08-10:
I found out that to make it work you have to remove the "desktop.ini" files that are created by google drive at the root of each folder.

danopernis, 2024-08-14:
@0ba-pia this seems like a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3817

Is either "google drive for desktop " or Windows itself marking the folder as "hidden" on FAT filesystem? If that is indeed the case, then I would argue that the printer behaves correctly.

I don't use Windows myself so I can't recommend any tools for you to check this. Perhaps right-clicking on the directory and selecting properties? Again, I am not a Windows user.

0ba-pia, 2024-08-14:
No, the folder isn't marked as "hidden" but it has special properties that are not recognized by the MK4.
Please note that the MK3S printers recognize those directories so it is a bug in the way MK4 reads the filesystem.

danopernis, 2024-08-14:
Looking at the code, there are two distinct flags in FAT filesystem: "hidden" and "system". If the entry (file or directory, doesn't matter) has any of those attributes set, [it is skipped](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/d8d5ec69ecd6725b5ea1369f9f34885728073551/src/buddy/filesystem_fatfs.cpp#L635) when walking the filesystem. This behavior seems to be quite old, even from MINI times before MK4. I am still not convinced if this is actually a bug, seems like those files are deliberately being ignored.

bkerler, 2024-08-30:
@0ba-pia Just to understand you right ... you have folders that aren't recognized but if you delete the desktop.ini, then they are recognized ? Could it be that the changes weren't uploaded by google drive / synced before ?

closed 2024-11-07 by github-actions[bot]