prusa3d/Prusa-Firmware-Buddy #4132: [BUG] Mk3.5 mmu3 fillament run out doesnt work

cozmo14047, 2024-08-11:
### Printer model

mk3.5 mmu3

### Firmware version

6.0.4

### Printing from...

prusa connect

### Describe the bug

When printing from prusa connect (untested without conenct) the fillament runout doesnt work. if i cut the fillament it will keep going with no fillament.

### How to reproduce

How to reproduce start a single color print using the mk3.5 template (as seen below).
cut the fillament and let it carry on printing fillament run out doesnt work.
[image]

### Expected behavior

Fillament runout should stop the print and ask for more fillament

Prusa-Support, 2024-08-17:
Thanks for bringing this to our attention but this doesn't seem to be a reproducible problem.

We may be missing something but this may sound like an unfortunate coincidence as our developers couldn't reproduce and study the problem, so please make sure to thoroughly double-check this case with the help of our Customer Support.
https://help.prusa3d.com/article/customer-support_2287

Chances are that the filament sensor settings have been unintentionally modified, or the filament sensor mechanism is slightly dirty, misplaced, or deformed.

Michele Moramarco
Prusa Research

cozmo14047, 2024-08-17:
> Thanks for bringing this to our attention but this doesn't seem to be a reproducible problem.
>
> We may be missing something but this may sound like an unfortunate coincidence as our developers [...]

Hi thanks for the reply, to note you did use the non mmu profile on prusa slicer so as not to enable the purge tower, i belive this is where the error lies. I can reproduce this so it is not a coincidence and everything is fully stock fillament sensors work fine. no loading issues and printer report correct state at all times.
would you like me to upload a 3mf of how i mean?
many thanks

closed 2024-10-28 by github-actions[bot]