prusa3d/Prusa-Firmware-Buddy #4137: [BUG] Nozzle does not cool down automatically after "Unload Filament"

AndraeCarotta, 2024-08-15:
### Printer model

MK4

### Firmware version

6.0.1

### Printing from...

USB / PrusaLink / PrusaConnect

### Describe the bug

- After unloading the filament, I would expect the nozzle to automatically cool down completely but instead the temperature is maintained after the process is completed
- In my opinion, this is a bug and the behavior is not intuitive. "Unload Filament" is only used if you do not want to continue printing and want to shut down the printer
- If you would then want to land new filament to move further you would select "Change Filament"

### How to reproduce

- just unloading filament

### Expected behavior

- after unloading the nozzle should cool down automatically

murk-sy, 2024-08-16:
Duplicate of
- #3691

See also pull request 
- https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3700

closed 2024-08-16 by danopernis

tlachmann, 2024-11-01:
I think. the behavior follows a intuitive logic... 
On unload filament the nozzle keeps heated and after you load a filament, the nozzle cools down directly to 170°C...