prusa3d/Prusa-Firmware-Buddy #4140: [BUG] Stopping Unload after Auto home crashes printer

murk-sy, 2024-08-16:
### Printer model

Mini

### Firmware version

6.0.4+14924

### Upgrades and modifications

Filament runout sensor

### Printing from...

n/a, direct control

### Describe the bug

While attempting to reproduce #4127, I discovered that a specific order of operation crashes the printer. I believe that issue may actually stem from the same basic problem, which would be related to axes being (mis)marked as homed or filament (mis)marked as unloaded, even though they may not be.

While at first I ran the procedure as unload - stop - auto home - unload - stop, the first unload-stop has not caused a crash and I managed to remove it from the reproduction steps.

### How to reproduce

1) Make sure filament is loaded and restart the printer
2) Run Control > Auto home
3) Run Filament > Unload filament
4) After X axis is homed to the right, press stop while the head is moving to the left
5) Printer becomes unresponsive (may happen after one press)

murk-sy, 2024-08-16:
Even though I made it I somehow forgot about this issue until now, but it's *very* similar
- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3830
Have not tested that recently

murk-sy, 2024-10-16:
Bumping but have not retested with current yet

CZDanol, 2024-10-16:
Hello, is this bug still present in the 6.2.0-alpha2 firmware?

murk-sy, 2024-10-17:
Reproducible on 6.2.0-alpha2, although it behaves slightly differently than I originally assumed - the issue was likely reproducible in the same way as it is now.

1) Make sure filament is loaded and restart the printer
2) Run Control > Auto home
3) Run Filament > Unload filament
4) ~~After X axis is homed to the right, press stop while the head is moving to the left~~
4a) X axis moves to the left to unload, press stop once it is stopped and heating
5) Printer becomes unresponsive (may happen after one press)

CZDanol, 2024-10-17:
Hey, thanks, I've managed to find the issue and it should be fixed in the next 6.2 candidate :)

Internal ticket: BFW-6193

murk-sy, 2025-03-07:
Appears to be resolved as of 6.2.2, closing

closed 2025-03-07 by murk-sy