prusa3d/Prusa-Firmware-Buddy #4141: [ENHANCEMENT] XL Firmware permanent Z Offset setting

lderossi, 2024-08-17:
### Printer model

XL

### Describe the enhancement

**Enhancement Request**
In XL firmware configuration on-device, add a ZOffset field that would default to zero. It should allow the same precision as the Live ZOffset (+/- 0.001 increments). This value would then be used for all prints, essentially pre-populating the Live ZOffset value at start-of-print.

**Justification**
The XL tool head calibration and pre-print bed level sampling do an excellent job obtaining a pretty good first layer every time. However there is room to dial this in even further when such precision is needed.

Across multiple freshly calibrated XL printers the same sliced gcode can still lay down a slightly different first layer as measured/observed in squish/extrusion width and first layer surface quality. For most prints this is inconsequential. However, in high precision features printed starting on the bed, even a small variance can cause fit issues (exaggerated by the fact that the first layer is, by design, squished onto the build plate a certain amount for adhesion). For whatever reason, tool head calibration and pre-print load cell bed sampling are unable to attain the proper precision to correct this.

This is reliably resolvable using the ZOffset in PrusaSlicer today, or adjusting live Z Offset at the start of print. However, in a print farm with multiple printers, this requires manual intervention at the printer (with a very short window, as Live ZOffset can't be done until extrusion begins), or files sliced for each XL with a custom ZOffset.

Please add a permanent ZOffset to the firmware, like the MK3s has, so this setting can be saved and used for all prints on each printer. This way a single gcode can be sent to any XL printer without user intervention and get expected effectively identical results.

bkerler, 2024-08-30:
As the printer uses unified bed leveling via loadcell by creating a leveling mesh, there is no such thing as z-offset for the XL/MK3.9/MK4 to my knowledge.

lderossi, 2024-08-30:
> As the printer uses unified bed leveling via loadcell by creating a leveling mesh, there is no such thing as z-offset similar to mk3/mk3s for the XL/MK3.9/MK4 to my knowledge.

As mentioned in the request, there is already the concept of a Live Z Offset in the XL firmware, access by pressing and holding the knob after printing begins.

The setting I'm after would simply set the default value of Live Z Offset to something other than zero.

After posting this, I stumbled across the manual page for a "fine adjustment to nozzle offset calibration" (involving a print with aiignment markers and manual addition to the nozzle offset numbers) but this was only for X/Y and only relative to head #1. 

Adding a calibration test for nozzle Z offset would be helpful, but I still see slight Zvariances between multiple XLs even on the first head so this wouldn't solve everything.

So far adjusting starting Z offset in the slicer solves my issues but forces one to slice a model once for each printer you plan to print on, which is very inefficient for a print farm.

bkerler, 2024-08-30:
@lderossi not sure how easy (or at all) this is achievable on the firmware side for the XL, but I will have a look into it

m00dawg, 2024-10-25:
Related: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3306

CZDanol, 2024-10-29:
Closing as a duplicate of #3306

closed 2024-10-29 by CZDanol