prusa3d/Prusa-Firmware-Buddy #4142: [BUG] Pause and Filament change command is ignored on Prusa Mini

dwbmb, 2024-08-18:
### Printer model

Mini+

### Firmware version

6.0.4

### Upgrades and modifications

None

### Printing from...

Prusa Connect

### Describe the bug

Same BUG as in my original post, that was closed without solution.

Still happening that change colour command is ignored in printer. I can print the same code over and over, even with printer turning off and on again in between with no help. This gcode is being printed without colour change. 
Gcode is generate din PrusaSlicer 2.8.0

[Knoflik_0.4n_0.1mm_PETG_MINI_1h34m.zip](https://github.com/user-attachments/files/16651505/Knoflik_0.4n_0.1mm_PETG_MINI_1h34m.zip)

BTW 
[Knoflik_0.4n_0.1mm_PETG_MINI_1h34m.zip](https://github.com/user-attachments/files/16651505/Knoflik_0.4n_0.1mm_PETG_MINI_1h34m.zip)
Is there any ways how to keep topic open before fixing bug so I do not have to issue the same bug over again?

### How to reproduce

Print this gcode on Prusa Mini+.

### Expected behavior

I expected printer to allow me filament colour change, it printed the whole model in one filament instead.

mix579, 2024-08-19:
Can you upload the 3mf file rather than the bgcode?

dwbmb, 2024-08-19:
Is that related somehow? Printed does not ever get into contact with 3mf, or does it?

mix579, 2024-08-19:
No, but to see if any settings are screwed up that may result in bad gcode, who wants to wade through a gcode file. But I looked at the gcode, and it definitely includes the M600, so yeah, looks like for some reason the printer ignores it.

danopernis, 2024-08-20:
I did try to reproduce this with file containing just the single `M600`. Curiously enough, result depends on the target hotend temperature. If the hotend is cold, `PREVENT_COLD_EXTRUSION` kicks in and `M600` does nothing (well, it does print "Hotend too cold" on the serial line). If the hotend is not cold, `M600` proceeds to change filament.

Now, your gcode contains custom filament-specific end gcode with `M104 S0` which sets the target temperature to zero, which might explain why nobody else is reporting this.

I am not entirely sure what the correct behaviour is here. I would argue that the M600 is working as intended. However,  maybe we should at least show warning in the user interface for users who are not using serial line to monitor/troubleshoot their prints.

P.S. Your issue #3784 was closed because it was inactive. There was a warning from the stale bot, suggesting to comment on the issue to prevent automatic closing. Nobody commented on the issue even after the warning, so the bot closed it after the timeout. To prevent issue from being closed, all it takes is to comment on it, which nobody cared to do.

dwbmb, 2024-08-20:
Hello, and thanks for your comment. Is there a way how to prevent this? I did never ever enter custom g-code. I am typical BFU, all I do is using PrusaSlicer with its built-in filament Profiles (Filament PM PETG in this case, in the previous report it was Filament PM PLA+). In fact I buy only filaments with built-in profiles in PrusaSlicer to avoid custom tweaking at all.

danopernis, 2024-08-20:
Hmm, interesting. Maybe there is something else which prevents the filament change then :confused:

Can we at least confirm if this is caused by the cold extrusion prevention? Can you please slice just some small cube (in order to save filament) and insert the pause somewhere near the first few layers? I would like to know what the target hotend temperature is during the filament change.

dwbmb, 2024-08-20:
Maybe I am not getting your point. I can slice a cube with a pause of course, but then what? Should I insert filament colour change right after and watch what happens?

closed 2025-12-16 by github-actions[bot]