prusa3d/Prusa-Firmware-Buddy #4146: [FEATURE REQUEST] Prusa XL Lighting Improvements

chrishilbert, 2024-08-20:
### Printer model

Prusa XL

### Describe the feature

Improve the XL lighting so it turns on/off automatically and has a top level menu button.

### Expected functionality

- Add an option to have the lights turn on with the start of a print and off at the completion of a print.

- Add the light on/off toggle to the top menu of the screen.  It's buried several button clicks in, bring it to the front during a print at minimum.  There's room on the screen.

legomind, 2024-09-09:
+1 for lighting toggle buttons

DoubleStrike, 2024-10-21:
I really like the idea of turning lights on/off when printing so I am not leaving LEDs turned on 24 hours a day attracting small bugs, etc.

CZDanol, 2024-10-22:
Related ticket: BFW-6168

Gefionious, 2024-11-03:
I support this request, when not printing, I would love a gcode command to turn the main side LED lights off and then while printing back on. Thank you!

spritefalcon, 2024-12-16:
+1. It makes the printer look cool but doesn't have to be lit 24/7. Could be an inbuilt menu like "Printer Lights" or "Lights During Print"

Gefionious, 2025-01-31:
+1 I support this request. I love the lights on when I am printing, but not on when I am not printing. A simple on/off Gcode command would be perfect. Or worst case a GPIO pin I can mount a switch on the printer. Thank you!

CZDanol, 2025-01-31:
Note: There is already a feature called "Side strip dimming" that dims the strips on user inactivity.

Gefionious, 2025-01-31:
> Note: There is already a feature called "Side strip dimming" that dims the strips after 120 s of user inactivity. It is configurable in Settings → User Interface.

Yes, thank you for bringing that up, but it is nowhere near sufficient, even the "dim" lights are too bright, brighter than any nightlight. An alternative would be to add a feature where we could select the dim value from 0-100. Then this timeout would work. I have no opinion on the easiest or best way to solve it, settings or Gcode, whichever is best and/or easier. It wouldn't matter for a print farm, but it does matter when I have the printer in my home office.

P-C-R, 2025-02-17:
This would be useful for the Core One too! Just give us M355 :-)

CZDanol, 2025-02-18:
> An alternative would be to add a feature where we could select the dim value from 0-100.

This has been implemented on the C1 in 6.3.0 (and will be on the XL, too, when 6.3 gets released for it)

CZDanol, 2025-02-18:
> Just give us M355 :-)
At the current state, it wouldn't be that easy. But we are working on rewriting the led controls - that should prepare the ground for this.

Gefionious, 2025-02-18:
> > An alternative would be to add a feature where we could select the dim value from 0-100.
> 
> This has been implemented on the C1 in 6.3.0 (and will be on the XL, too, when 6.3 gets released for it)

Fantastic, thank you!

mootweaker, 2025-08-11:
This would be useful. My Core One printer is close to a window that faces the street. I don't want it to attract attention at night.

foorschtbar, 2025-12-04:
> M355

I would like to bring `M355` this to your attention once again. I print with OctoPrint and need an option to turn the light on and off via OctoPrint.