prusa3d/Prusa-Firmware-Buddy #4148: [FEATURE REQUEST] Prusa Connect Option to set print queue to automatic so you don't need to click "Set Ready" between prints

JTuukka, 2024-08-21:
### Printer model

All Prusa Connect printers

### Describe the feature

A setting in the Prusa Connect settings menu that sets a certain printer into an "always ready" mode where the user does not have to press the "Set Ready" button before starting a new print from the print queue.

### Expected functionality

I'm doing mass production on tall and narrows parts that detach very easily when the bed is cold. I put my printer on an angled surface and added a couple of simple lines of gcode in the slicer to the end which waits for the hotend and bed to cool down and then rams the part off the bed down into a tray. Therefore, immediately after the print is ready the bed is also clear and ready for the next part to be printed. Currently I have to open the gcode in notepad and copy the same print multiple times over to simulate automatic printing which is a very non-elegant solution.

I would like to be able to set Prusa Connect to automatically start the next print on the queue without user input.

Ronin0000, 2024-08-21:
I would recommend to put this in the Prusa Connect Feature Request, as it would have a greater chance of being added there.
[image]

JTuukka, 2024-08-21:
Where can I find this "Prusa Connect Feature Request" forum? I can't find any forum with matching posts and an upvote/downvote system

Ronin0000, 2024-08-21:
It should be shown in the bottom right corner of Prusa Connect as a icon of
a question mark. [image: image.png]

JTuukka, 2024-08-21:
Oh I see I completely missed that, thought it was somewhere on the internet. Thanks, I'll definitely post there!

bkerler, 2024-08-30:
@JTuukka please close this issue as it is Prusa Connect and not Printer Firmware related. Thanks :)

closed 2024-08-31 by JTuukka