prusa3d/Prusa-Firmware-Buddy #4149: [BUG] Touchscreen control for items in "temperature" menu doesn't work

me25542, 2024-08-22:
### Printer model

MK4

### Firmware version

6.1.2+7894

### Upgrades and modifications

None

### Printing from...

N/A

### Describe the bug

Controlling nozzle temperature, bed temperature, or print fan speed doesn't work when using the on-screen keyboard.
When, for example, setting the print fan speed to 100%, I typed in "100", pressed enter, and the menu _said_ 100% for the print fan speed. However, the print fan doesn't actually turn on.
When setting the temperature via the scroll knob, the print fan turns on as expected.
The same issue appears when setting the nozzle temperature and for setting print bed temperature; the menu _shows_ that the temperature is updated, but the footer does not, and the nozzle / bed doesn't actually turn on.
"cooldown" still functions as expected when selected via the touchscreen.

-Note-
Resetting the printer does not appear to correct this issue

### How to reproduce

Go to main menu - control - temperature (using touchscreen control), and set nozzle temperature, bed temperature, or print fan speed to some value other than their current value using the on-screen keyboard.

### Expected behavior

Users should be able to set the nozzle temperature, bed temperature, and print fan speed using the on-screen keyboard.

H0Z0, 2024-08-22:
I can confirm. The set values ​​are displayed, but nothing happens. After setting the values, exiting the menu and then returning, the values ​​are also deleted.

neofir3, 2024-08-22:
I can confirm the problem on my MK4 with 6.1.2
But if you enter 100% on the touch display and confirm it AND then press the normal button, the 100% is accepted and the fan starts.
But normally it should work immediately after touch input, just like with the nozzle and board.
But I was also surprised that you have to touch directly at the back where the number is.
I would have thought it would be enough to just touch the corresponding line. But if you touch the front area, the numeric input field does not appear.

neofir3, 2024-08-22:
I can confirm the problem with my MK4 with 6.1.2
But if you confirm the touch input with the normal button, the respective value is adopted.
What also surprises me is that the touch input only works if you touch the right-hand area where the numbers are.
If you touch the front of the line, the touch input window does not open at all.
I would have thought that it would be enough if you just touched the corresponding line. But if you touch the front area, the number input field does not appear.

danopernis, 2024-08-23:
Thanks for reporting @me25542 @H0Z0 @neofir3 I can confirm this bug is present. I have created internal ticket for further investigation.

@neofir3 numeric keyboard only appearing after clicking on the numeric value on the right side of the menu item is intended behaviour, not a bug. This was intended to prevent misclicks while scrolling in the menu. Perhaps this is misleading to some people though. Feel free to open separate feature request if you think this should be changed.

H0Z0, 2024-09-04:
Confirmed, working well.

closed 2024-09-06 by danopernis