prusa3d/Prusa-Firmware-Buddy #4150: [BUG] Faster Z-Homing speed causes Nozzle pushing down heatbed

Iffy0o, 2024-08-22:
### Printer model

MK4

### Firmware version

6.1.0+7864

### Upgrades and modifications

None

### Printing from...

USB

### Describe the bug

When I installed the FW V6.1.0+7864 onto my MK4 i recognized the homing procedure is quicker in general, but the nozzle probes to hard into the heat bed when homing the Z axis. 

Commit 4243c3e "Faster Z-homing speed" is probably the cause for this.
in include/marlin/Configuration_MK4.h
Before:
`#define HOMING_FEEDRATE_Z (8 * 60)`
Since 6.1.0:
`#define HOMING_FEEDRATE_Z 1150`

Faster speed also results in later end stop recognition because of more traveled distance per time. And pushing down on the edge of the printbed will surely degrade the mounting faster.

### How to reproduce

Use FW >= V6.1.0+7864 on the MK4 and auto home in the menue

### Expected behavior

No heat bed bending when homing the z axis

### Files

https://github.com/user-attachments/assets/90e94cee-a2fa-4219-b6a9-662ea58eca48

H0Z0, 2024-08-27:
Bed "bending" was always present even with older fw at lower HOMING_FEEDRATE_Z.

https://github.com/user-attachments/assets/53aa332d-1c51-4c9e-8ed0-fc035d112ef4

danopernis, 2024-09-16:
Hi folks, I consulted this with my colleagues from hardware team and the testing department and they assured me that this is not an issue. The mechanical force is not large enough to negatively affect the bearings.

closed 2024-09-16 by danopernis